nornagon/saxi #1: Track AxiDraw status & automatically reconnect

kylestetz, 2019-01-20:
Unless I'm mistaken the server only connects to the AxiDraw once at startup, so e.g. canceling a plot and unplugging the power (to re-zero the machine) drops the connection and requires restarting the server and refreshing the UI.

It would be nice to 1) automatically reconnect, maybe by polling while disconnected, and 2) display the connection status in Saxi, so that the UI can be a little more resilient.

[image]

nornagon, 2019-01-20:
You shouldn't need to unplug the power to re-zero the machine (that's what the "limp" button does), but I agree we should auto-reconnect to the serial port.

Also, it's kind of a separate issue, but the UI currently only opens the socket to the server one time, so if you close your laptop & re-open it then you'll be disconnected from the axidraw & have to refresh to reconnect & get status. ("plot" and "cancel" will still work as they issue POST requests, but penup/pendown won't.)

I'll open a new issue for the UI reconnection.

closed 2019-03-16 by nornagon

nornagon, 2019-03-16:
@kylestetz this could use some nice UI to indicate the connection status. As is there's no way in the UI you can know if the robot's connected or not other than by hovering the 'saxi' logo and seeing if it has a tooltip like `/dev/tty.usbmodem14421`

commit 9c6348c6 (2019-03-16): feat: reconnect to EBB when disconnected

Closes #1