nornagon/saxi #18: Windows Compatibility

JanWerder, 2019-03-15:
I would be great if saxi would work on Windows as well

nornagon, 2019-03-15:
I'm not aware of any issues on Windows. What problem are you running into?

JanWerder, 2019-03-15:
I installed saxi over npm, but it goes into simulation mode as soon as I run the command.

nornagon, 2019-03-15:
Ah, I see. Is the AxiDraw plugged in when you start saxi?

If saxi still enters simulation mode on startup even if the AxiDraw is connected, can you try running this and paste the output:

```
$ node -e "require('serialport').list().then((ps) => console.log(JSON.stringify(ps)))"
```

JanWerder, 2019-03-15:
Okay, it says:
```Error: Cannot find module 'serialport'```
After installing serialport via npm, saxy still doesn't find the Axidraw, but I can give you the output of the serial port listing:

```[{"comName":"COM1","manufacturer":"(Standardanschlusstypen)","pnpId":"ACPI\\PNP0501\\0"},{"comName":"COM6","manufacturer":"SchmalzHaus LLC","serialNumber":"7&195cdef5&0&3","pnpId":"USB\\VID_04D8&PID_FD92\\7&195CDEF5&0&3","locationId":"Port_#0003.Hub_#0003","vendorId":"04D8","productId":"FD92"}]```

nornagon, 2019-03-15:
Try running:

```
saxi -d COM6
```

It looks like the manufacturer on your board is slightly different from mine, which is interfering with the detection: https://github.com/nornagon/saxi/blob/master/src/ebb.ts#L17

closed 2019-03-15 by nornagon

JanWerder, 2019-03-15:
Thanks, that fixed the detection. 😊
PS. Is the fix automatically pushed to npm?

nornagon, 2019-03-15:
Nope, it's not automatic. I'll push a new release in the next day or two though!

nornagon, 2019-03-18:
(in case you didn't already see, v0.5.0 should have the fix. happy plotting!)

Makio64, 2020-04-18:
It should be good to add it to the documentation or enable auto-detection on other COM, personnaly it works for me with windows "saxi -d COM5"
Thanks for this tool!

nornagon, 2020-04-18:
@Makio64 Is autodetection not working for you? Can you run the command in https://github.com/nornagon/saxi/issues/18#issuecomment-473395460 and paste the output here?

Makio64, 2020-04-18:
The autodetection didnt work, windows 10 razer pro 17" :
Here the output : 
 `[{"path":"COM5","manufacturer":"Microsoft","serialNumber":"5&68c6c1f&0&1","pnpId":"USB\\VID_04D8&PID_FD92\\5&68C6C1F&0&1","locationId":"Port_#0001.Hub_#0002","vendorId":"04D8","productId":"FD92"}]`

nornagon, 2020-04-18:
Oh hm. Are you using a USB hub? It's weird that the manufacturer name doesn't show up for you.

It looks like the vendor id / product id are showing up though, which is [how the axidraw cli](https://github.com/evil-mad/axidraw/blob/90dcd9416d04ab487111eb6b0d2bb983d85e8183/cli/pyaxidraw/ebb_serial.py#L67) finds the device I guess. I'll add that logic in to saxi!

nornagon, 2020-04-18:
Should be fixed by f89cb5547842fb50b9acc582a2713eb0007deb13, I'll make a release later today.

Jip-Hop, 2020-09-02:
Looking forward to the next release :) Installed from NPM today but it wouldn't find the connected AxiDraw. `saxi -d COM3` works though.

nornagon, 2020-09-10:
@Jip-Hop can you try the steps in this comment? https://github.com/nornagon/saxi/issues/18#issuecomment-473395460

Jip-Hop, 2020-09-11:
`[{"path":"COM1","manufacturer":"(Standaardpoorttypen)","pnpId":"ACPI\\PNP0501\\1"},{"path":"COM3","manufacturer":"Microsoft","serialNumber":"6&1ba46697&0&5","pnpId":"USB\\VID_04D8&PID_FD92\\6&1BA46697&0&5","locationId":"Port_#0005.Hub_#0004","vendorId":"04D8","productId":"FD92"}]`

I connected the AxiDraw directly to a USB port on the back of my Dell Optiplex 990 PC. So no external USB hub is used.

nornagon, 2020-09-11:
@Jip-Hop great, thanks! It looks like https://github.com/nornagon/saxi/commit/67429dfc5a4a00614ef9bcf6e6460e9d15427b7c will fix this issue for you :) I thought I'd already released that, but it looks like I haven't! Will get to a release soon.

nornagon, 2020-09-16:
@Jip-Hop v0.10.0 was just released which should include this fix.

Jip-Hop, 2020-09-20:
Confirmed working :)

commit e7ff5b0d (2019-03-15): fix: accept "SchmalzHaus LLC" as a manufacturer

Closes #18