nornagon/saxi #23: Automated releases

blakedietz, 2019-03-22:
Would you Ben interested in automating releases along with ci/cd?

We could set up ci/cd on circleci. Then from there use semantic release with commitizen to make reasrs to npm as simple as pushing to master.

I’m making an issue before pr because I would need to work with you to set up the circleci accounts. I would need two things to help out:
- a gh token with write access to the repo
- an npm token to push to the project. 

I could just implement and or and give you the instructions for setting up the variables too.

nornagon, 2019-03-22:
Hey thanks for offering!! This seems interesting but possibly overkill at this stage?

1. There's no test suite currently, so there's no way to tell if master is completely broken. I'd say that some kind of (even fairly minimal) test suite is a hard requirement before doing automated releases.
2. Pre-1.0.0, I'm not sure I want to automatically release on merge to `master`? Sometimes there are intermediate states that are worth pushing to `master` but which aren't really release-ready yet. It might make sense to trigger releases off of manually-created git tags?

I wrote a little script (in `tools/release.js`) that automatically updates the version in package.json & creates a git tag. It's just a little bit of automation but it's been helpful :)

Would love to get tests going on circleci though! I made a new issue for that: #24

blakedietz, 2019-03-28:
This makes sense. I'll start looking at #24.

jedahan, 2023-09-06:
I think this can be closed?

alexrudd2, 2025-01-24:
I'm happy with the manual release process.

closed 2025-01-24 by alexrudd2