nornagon/saxi #25: Feature request: Return to home

blakedietz, 2019-03-23:
Would it be possible to have a return to home function?

nornagon, 2019-03-23:
Hm, I'm not sure I understand your use case. "Cancel" disengages the motors, which lets you just push the robot back to (0,0). Further, if you move the head by hand at all then a "Return Home" button could very easily run into the end of the axis.

Could you explain a little more about what you're envisioning here & what need this would address?

blakedietz, 2019-03-28:
Actually, let's close this issue. I'll reopen if I end up changing my mind.

I misunderstood and thought you would be able to add a return to home function for the plotter so if you cancel a print, instead of disengaging the plotter would return home.

closed 2019-03-28 by blakedietz

nornagon, 2019-03-28:
Hm, I think it might be better not to automatically return home when cancelling a plot. It could be that the user is cancelling the plot due to a pen issue (e.g. not clamped tightly enough), & cancelling the plot could end up dragging the pen across the paper.

I can imagine a "return home" feature being useful if you were using a point other than the physical minimum limit of the AxiDraw's travel.