nornagon/saxi #31: Bug: Pen stops during long prints

blakedietz, 2019-03-28:
## What are the steps to reproduce this issue

I'll generate an svg that causes this.

### Images or video

I'll record a video to show what happens.

## What happens

After a long print the axidraw machine will stop mid print for no reason, however saxi continues forward.

## What were you expecting to happen

I expect saxi to continue drawing until the pause button is pushed or the print is finished or cancelled.

## Debug information

None right now

## Any other comments

I know that there isn't much info here. This has been happening for nearly every print I've done. Intrestingly if you just click the pen down button (even though the pen is already down) the plot will continue without a hitch. Eventually plotter catches up with the saxi crosshairs.

## What versions of software are you using

This was pre pause button release I still need to snag the number.

### Operating System

macOs: mojave
node: v11.5.0

nornagon, 2019-03-28:
What version of saxi?

blakedietz, 2019-03-28:
Based on date of releases and when I discovered saxi, I'm guessing v0.6.0.

blakedietz, 2019-03-28:
Actually I just checked on the machine I was plotting on and it was v0.6.1.

nornagon, 2019-03-30:
Thanks for the info. One more thing, what does `saxi --firmware-version` tell you?

blakedietz, 2019-04-09:
^ I'll snag this and respond. Also I got a video the other day. I don't have any log info though.

nornagon, 2019-09-15:
@blakedietz did you end up being able to reproduce this? I still have never personally observed this issue sadly :( (or happily, I suppose. But sadly from the perspective of being able to fix it.)

blakedietz, 2019-09-15:
Eh, let’s close it for now. 

Blake Dietz
(208) 577-1368

blakedietz, 2020-05-29:
This is still happening. I have an svg I can send you if you'd like to try it out. I'll add some os info for repro.