nornagon/saxi #33: Installing Saxi

Ohmarinus, 2019-07-15:
Hi, I'm trying to get this to work but it keeps giving an error:

`sudo npm i -g saxi
/usr/local/bin/saxi -> /usr/local/lib/node_modules/saxi/cli.js

> @serialport/bindings@2.0.8 install /usr/local/lib/node_modules/saxi/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

shell-init: error retrieving current directory: getcwd: cannot access parent directories: Permission denied
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: Permission denied
sh: prebuild-install: command not found
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: Permission denied
sh: node-gyp: command not found
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/marinusdebeer/.npm/_logs/2019-07-15T10_47_15_625Z-debug.log
Marinuss-MacBook-Pro-9:saxi-master marinusdebeer$ saxi
-bash: saxi: command not found
`

What can I do to fix this? I keep running into these kind of errors. I'm working on a Macbook with Yosemite.

Ohmarinus, 2019-07-15:
I have found a fix for this, it seems to be that if users get these issues they should first set r/w access to the saxi folder in /usr/local/lib/node_modules

And then use this command:
--unsafe-perm

So:
$ npm i -g saxi --unsafe-perm

nornagon, 2019-07-15:
you shouldn't need `--unsafe-perm` if you set the permissions to 0755. Here's what the permissions look like on my machine:
```
$ ls -l /usr/local/lib |grep node_modules
drwxr-xr-x    9 nornagon  staff   288B Jul 13 09:50 node_modules/
```

Glad you found a workaround though! Closing this issue as it's not a saxi issue but rather a configuration issue on your machine.

closed 2019-07-15 by nornagon