nornagon/saxi #34: TypeError: Argument 1 of SVGPoint.matrixTransform is not an object

fitzgen, 2019-08-17:
### Test Case

`triangle.svg`:

```svg
<svg viewBox="0 0 70 70" xmlns="http://www.w3.org/2000/svg">
    <path d="M10,10 L60,20 L25,60 z"/>
</svg>
```

### Steps to Reproduce

* Start `saxi`
* Drag `triangle.svg` into saxi

### Expected Results

Saxi plans how to plot `triangle.svg`.

### Actual Results

The UI gets stuck in the "I see you are dragging a file over here, but haven't yet dropped it" state, and becomes unresponsive.

In the console, there is this error:

```
TypeError: Argument 1 of SVGPoint.matrixTransform is not an object. svg-to-paths.js:79
    o svg-to-paths.js:79
    flattenSVG svg-to-paths.js:87
    q ui.tsx:577
    q ui.tsx:584
    onload ui.tsx:504
```

### Environment

```
$ saxi --version
0.7.0
$ node --version
v10.16.0
```

-----------------

Any idea what's going on? Any other information I can provide? Thanks!

fitzgen, 2019-08-17:
It seems like a call to `getCTM` on an svg path is returning `null` and the code is not prepared to handle that, and is passing the `null` to `transformMatrix`:

[image]

fitzgen, 2019-08-17:
It seems like `getCTM` only returns `null` when the svg element is not in the document: https://www.w3.org/TR/SVG/types.html#__svg__SVGGraphicsElement__getCTM

nornagon, 2019-08-17:
I suspect this is a Chrome/Firefox difference. I'll take a look when I'm back from vacation unless you want to take a stab at a PR! The code's over in [flatten-svg](https://github.com/nornagon/flatten-svg/blob/master/src/svg-to-paths.ts#L98). Seems like a quick fix would be to add the SVG to the document and then ensure its removal before the browser attempts to render it with try/finally.

fitzgen, 2019-08-18:
Hm ok so I implemented a force-the-svg-to-be-connected-to-the-DOM and `getCTM` is still returning `null` so something funky is going on here.

Not sure exactly how this is being used, but maybe using an identity matrix in `getCTM` returns `null` will work?

fitzgen, 2019-08-18:
> Not sure exactly how this is being used, but maybe using an identity matrix in `getCTM` returns `null` will work?

This seems to work! Will whip up a PR.

fitzgen, 2019-08-18:
Fix is over in https://github.com/nornagon/flatten-svg/pull/2

closed 2019-08-19 by nornagon