nornagon/saxi #37: Have drawn lines change color

zoso95, 2019-09-13:
If you had the lines that were drawn change color that would give a pretty good indicator of progress

nornagon, 2019-09-15:
Yes, this would be rad! I had a go at this a while ago and ran into performance issues with redrawing complex SVGs, but if we can find a way to do it performantly that would be great.

(Perhaps we could split up the drawing into many small layers of ~1000 lines each, then we'd only be redrawing at maximum two layers per frame...? Not sure if the many layers would be better or worse, though.)

Makio64, 2020-04-18:
I think it might be done with gsap without too much extra work ?
https://greensock.com/docs/v3/Plugins/DrawSVGPlugin

nornagon, 2020-04-18:
The problem here is with the browser's own redraw rate of complex SVGs. It doesn't matter which library is used to trigger the relevant redraws, it's still going to be the browser redrawing the SVG at the end of the day.