nornagon/saxi #38: Saxi sometimes collapses lines poorly

zoso95, 2019-10-11:
The path joining radius should consider the line length, because for long lines a small change in the starting location can change the geometry by a decent amount

nornagon, 2019-10-15:
Hey, thanks for the report! Yeah, I agree the algorithm could be improved. It would be super helpful if you could provide an SVG that exhibited the issue that I could test with.

Did you find the option to turn the path-joining radius to zero? That should at least let you plot the image.

zoso95, 2019-10-21:
Oh yeah i found it after the fact. It was when I was doing a cross hatching image where the lines were very close together to create a dark black

You can see it in this image where lines will be joined at the base and the create a lighter apperance 

https://www.instagram.com/p/B3sG__wnI-b/