nornagon/saxi #39: SVG with different colored strokes not being broken apart in to layers

lostPixels, 2019-12-04:
I have an SVG with 5 different colored stroked and 5 different layers. I was under the assumption that Saxi would allow me to plot each layer one at a time, so that I can use different color pens. 

I noticed in the dev console that I'm getting this output:
``` js
posting to worker: 42.85791015625ms
background-planner.ts:29 sorting paths: 4637.9599609375ms
background-planner.ts:39 joining nearby paths: 123.255859375ms
background-planner.ts:61 planning pen motions: 99.2080078125ms
background-planner.ts:11 serializing: 9.08203125ms
ui.tsx:225 deserializing: 217.10302734375ms
VM96 backend.js:1 Uncaught RangeError: Maximum call stack size exceeded
```

I'm not sure if this is related or not. The SVG is around 1.5mb.

nornagon, 2019-12-04:
Oh dear, that's not the expected behavior at all! Would it be possible for you to link me the SVG? (If it's something you'd rather not share publicly, you can also email it to me at nornagon@nornagon.net)

lostPixels, 2019-12-11:
Hello, just wanted to validate that you received the test SVG. No rush on the fix, thanks! Once you commit I'll take a look at your code too and maybe I can jump in contribute to future features.

nornagon, 2019-12-11:
Hey, yep, thanks for checking—I did receive it, just been busy with other
projects. I’ll get to it!

nornagon, 2019-12-13:
Ok, I took a look at the SVG you sent. Unfortunately, it seems that the tool you're using to generate these SVGs uses CSS to apply colors. Saxi is quite stupid about this and doesn't understand CSS colors; it can only understand colors that are applied as a `stroke` attribute on the elements themselves.

e.g. this will work:

```svg
<g>
  <path d="M0 0 L10 10" stroke="red" />
  <path d="M10 0 L0 10" stroke="blue" />
</g>
```

But this will not:
```
<style>
.a { stroke: red; }
.b { stroke: blue; }
</style>
<g>
  <path d="M0 0 L10 10" class="a" />
  <path d="M10 0 L0 10" class="b" />
</g>
```

There's no particularly good reason I'm aware of that saxi couldn't be made to understand the CSS colors too, it's just that it wasn't easy to figure out how to do that when I was writing the code, and the SVGs that I happened to be plotting at the time used the stroke attribute on individual elements.

The code that extracts color information is here: https://github.com/nornagon/flatten-svg/blob/master/src/svg-to-paths.ts#L116

nornagon, 2019-12-14:
NB, I've just released v0.8.1, which has some improvements in this area, but still unfortunately won't fix this specific problem.

These now work (they didn't previously):

The `stroke` attribute on an ancestor element will now work to set the layer for a path:
```svg
<g stroke="green">
  <path d="..." />
  <path d="..." />
</g>
```

The `style` attribute is now respected, both on the element itself and on an ancestor:
```svg
<g style="stroke: green">
  <path d="..." />
  <path d="..." style="stroke: red" />
</g>
```

lostPixels, 2019-12-16:
It turns out that exporting from Illustrator has an open to use CSS or strokes. It's hidden behind the Advanced Options > CSS Properties of the export dialog. Setting it to "Presentation Attributes" will save an SVG that Saxi successfully separates based on stroke color. 

Thanks for your assistance with this! Now all my artwork is way more colorful!

closed 2019-12-16 by lostPixels

nornagon, 2019-12-16:
Fantastic! Would love to see a photo if you want to link it here :)

lostPixels, 2019-12-16:
For sure! Without Saxi this would have been way more difficult to plot. https://i.imgur.com/ueIj3MA.jpg

nornagon, 2019-12-16:
🤩wow, that's gorgeous! So glad you got it working :) Thanks for sharing!