nornagon/saxi #40: Saxi doesn't recognise some multi-layer files

abey79, 2020-02-11:
I have an example of multi-layer SVG file for which SAXI will not show the layer selector. How can I plot a single layer of this file?

Here is the file:
[patterns_8x.svg.zip](https://github.com/nornagon/saxi/files/4187447/patterns_8x.svg.zip)

nornagon, 2020-02-11:
Saxi uses stroke color, not svg layers (groups) to split layers apart. Try making each layer a different stroke color?

abey79, 2020-02-11:
Oh ok. Thanks for the quick answer -- this solves my immediate issue. That said, I will have to vary layer colours in [vpype](/abey79/vpype/)'s export so make its output directly compatible with saxi. Is colour the only attribute that's being considered or are there others? Is it ok if colour is set at group level (as opposed to individual path?

nornagon, 2020-02-12:
yep, setting the stroke color on the group works since saxi [v0.8.1](https://github.com/nornagon/saxi/releases/tag/v0.8.1).

nornagon, 2020-02-19:
Spent some time thinking about this tonight. I'm uncertain how to handle the two separate modes of layering, here are some possibilities:
1. Stop supporting layer-by-stroke, switch to layer-by-group-id completely. Layer-by-stroke is a bit of a hack anyway.
2. Detect whether there exists a `<g id="...">` anywhere in the SVG, and if so, use layer-by-group-id. Otherwise, use layer-by-stroke.
3. The inverse of (2), use layer-by-group-id when layer-by-stroke would produce exactly 1 layer (i.e. there are no stroke-separated layers).
4. Allow toggling between grouping modes in the UI.

Currently I'm tending towards (3), though it might be good to have an override. Perhaps if it detects that either stroke-based or `<g>`-based methods would both produce >1 layer, it could allow toggling between them?

closed 2020-02-19 by nornagon

nornagon, 2020-02-19:
Give 0.9.0 a whirl, I went with (3) + (4) :)

abey79, 2020-02-19:
Looks great, will try it out ASAP. Thanks for taking the time!

commit 28c86b71 (2020-02-19): feat: layer by group id when no stroke info

Fixes #40.