nornagon/saxi #41: Add possibility to disable SVG scaling

abey79, 2020-02-12:
Currently, the only way to control how the SVG is scaled on the chosen page format is to change the margin setting. This works well in many cases and is a nice default behaviour, but is inconvenient for cases where careful scaling has been done beforehand. Here are a few examples I can think of:

- plots with filled polygons where hatching pitch must be precisely controlled
- technical plots where final, physical scale is important
- small plots in a specific place on the page
- etc.

nornagon, 2020-02-12:
Yeah, I agree this would be handy! It should be fairly easy to add.

As a sort-of clunky solution you can use today:
1. Create an SVG which includes a rectangle at the exact ratio of the page. e.g. if you're targeting A4, create an SVG with dimensions in the ratio of 210:297.
2. Add a rectangle from 0,0 to 210,297 with `stroke="red"`. Draw your main plot with `stroke="black"`.
3. When plotting in saxi, set margins to 0 and select the "black" layer. saxi will rescale the plot so the _entire_ plot (including hidden layers) will fit in the page. Since the red layer is the full size of the page, it will be scaled at 1:1 exactly.

example svg:

```xml
<svg xmlns="http://www.w3.org/2000/svg" width="297" height="210" viewBox="0 0 297 210">
<rect x="0" y="0" width="297" height="210" stroke="red" fill="transparent" />
<g stroke="black" fill="transparent">
  <rect x="10" y="10" width="10" height="10" />
</g>
</svg>
```

abey79, 2020-02-13:
Makes sense, thanks for the workaround. I'll probably use it!

commit 30789d02 (2020-02-19): feat: allow disabling fit-to-page

Fixes #41
closed 2020-02-19 by nornagon