nornagon/saxi #42: Bug: disabling 'fit page' does not respect SVG scale

abey79, 2020-02-24:
Consider the following  SVG as seen in InkScape:
[image]
Once loaded in saxi, the scale is clearly not respected:
[image]

Here is the SVG file: [vader_10cm_opt.svg.zip](https://github.com/nornagon/saxi/files/4245587/vader_10cm_opt.svg.zip)

It may be that the `width` and `height` attributes of the `svg` tag are not considered. (Note that in this present SVG, the `viewBox` attribute is omitted, but should be honoured as well.)

abey79, 2020-02-24:
FWIW, [here is](https://github.com/abey79/vpype/blob/09e0ef5fb298556fc6fecc8381042b911d0fd203/vpype/read.py#L43) how I do it in vpype. I'm using SVG pixels (1/96th of an inch) internally, thus the scale factor of 1 when `viewBox` is omitted. You'll need some further scaling for whatever units you are using internally.

nornagon, 2020-04-17:
Should be fixed by 05b0b7b, will release shortly. Not sure yet if it correctly handles viewbox, but your test svg seems to work correctly at least!

commit 05b0b7b6 (2020-04-17): fix: when not fitting to page, scale to correct units

Fixes #42.
closed 2020-04-17 by nornagon