nornagon/saxi #44: Long path get Axidraw jerky and slow

TarGz, 2020-03-16:
When I use a path using a very long and continuous line get Axidraw to be very jerky and slow while printing. 
It make a nasty and unusual sound. The saxi preview move normally but the actual printer get far behind in term of progress. I made some test below.

## TEST FILES 
I generated some test file : 
TEST A use each line are connected 
TEST B lines are not connected 
FULL all line are drawn
SKIP only a few lines are drawn.
[TEST_A_FULL](https://www.dropbox.com/s/mgrog41y9nkfsyk/TEST_A_FULL.svg?dl=0) -> FAIL 
[TEST_A_SKIP](https://www.dropbox.com/s/snemgp0sv0adxkk/TEST_A_SKIP.svg?dl=0) -> OK
[TEST_B_FULL](https://www.dropbox.com/s/gk5sy9vpnozbm6f/TEST_B_FULL.svg?dl=0) -> OK
[TEST_B_SKIP](https://www.dropbox.com/s/gyjsr9xql5uw1tb/TEST_B_SKIP.svg?dl=0) -> OK

## CONCLUSION
It looks like the fact that the file TEST_A_SKIP means that connected lines are OK 
meaning that TEST_A_FULL fail because it has a very long continuous line  

## SOFTWARE
saxi version : 0.9.1
axidraw firtmaware version : 2.6.5

go2dev, 2020-09-06:
Similar experience with long paths; machine moves much slower than the on-screen indicator and the arm jerks around causing the pen stroke to be very uneven. 

In some instances, applying the 'simplify path' tool to long paths in Inkscape will allow the line to be plotted - but this isn't always true. Using 'simplify path' also appreciably changes the visual outcome of the plot so is not a desirable workaround in most cases.

My test files and a video can be found in this dropbox folder:

https://www.dropbox.com/sh/2asofauno33qi2e/AABPNq37h2rRkJPWkbz_ASyxa?dl=0

In this test, both the simplified and original paths fail to plot correctly using saxi - the file being run in the video is testPlot.svg (which simply places longPlot.svg and longPlot_simplified.svg onto a single page). 

The files plot correctly from within inkscape.

### SOFTWARE ###
saxi version : 0.9.2 (on Windows 10 [intel NUC] and rasbian [Raspberry Pi 3B+] - same result on both)
axidraw firtmaware version : 2.6.5 and 2.5.1 (same results)

haschdl, 2022-02-25:
I'd like to share a similar case. The noises my Axidraw made while plotting my file were new to me, and I had to stop the plotting mid-way as it didn't sound right. 
I have another issue where the plotter seems to lose its reference point and half-way into plotting, it seems like it looses it's zero position, and it starts drawing all over the place, including off-limits. Here is a picture of a ruined plot, along with how it was supposed to look (screenshot from inkscape).
It is very unfortunate because the tool can be a tremendous time-saver, but right now it feels unreliable, wasting time (and some good paper).

@go2dev  @TarGz  did you find a good workflow to solve your issue with Saxi?

Result:
[image]

[image]

* Saxi 0.14.0 
* Axidraw: SE/A3 
* Firmware: EBBv13_and_above EB Firmware Version 2.7.0

go2dev, 2022-03-05:
Only to split the line in inkscape so it is not a single long line and then plot them in two (or more) different files. It works for some plots but the pen lift messes up some images as the pen/ink bleeds at the join so it isn't ideal

alexrudd2, 2025-01-20:
@go2dev @haschdl @TarGz Would you mind retesting on `0.16.0`?  It's using the fork https://github.com/alexrudd2/saxi and has a plotting fix that has helped other people.