nornagon/saxi #47: Pen up/down not working on Raspberry Pi

duncangeere, 2020-04-19:
I'm running Saxi on a Raspberry Pi 2 model B, and everything seems to work fine except that the software can't seem to move the pen up or down. The same Pi controls the Axidraw fine through Inkscape, so it's not a hardware issue.

The pen up/down buttons in the UI don't do anything, and during a plot the pen remains at the up position (though the arm moves around the plot as expected). No error messages are displayed.

Any idea what might be going on?

nornagon, 2020-04-19:
Oh no, that's weird! saxi does use a slightly different command to raise/lower the pen: we use the [S2](https://github.com/nornagon/saxi/blob/f89cb55/src/ebb.ts#L123) command, whereas axidraw's own firmware uses [SP](https://github.com/evil-mad/axidraw/blob/90dcd9416d04ab487111eb6b0d2bb983d85e8183/cli/pyaxidraw/ebb_motion.py#L213). It could potentially be an issue related to that difference?

Here's some debugging steps you can try:

1. Run `saxi --firmware-version` and report back the firmware version of your Axidraw. I've tested saxi mostly on my own device which is running version 2.5.3. If you're running a significantly older version, you can [try upgrading](https://github.com/evil-mad/EggBot/tree/master/EBB_firmware). If you're running a newer version, it may be that the S2 command has changed and I'll need to update saxi to handle it.
2. It could be that you got unlucky and your servo motor broke in between when you tried with the axidraw plugin and when you tried saxi. Can you verify that that's not the case by switching back to the axidraw plugin and trying that again? Just to verify that the servo's definitely still working.

duncangeere, 2020-04-19:
Firmware version is 2.6.5. Looks like that might be the issue?

I get a GUI on the Pi right now to test Inkscape there, but pen up/down still works fine when controlled from Inkscape on my Mac.

nornagon, 2020-04-19:
Ok, cool, looks like maybe a firmware issue then. I see there have been some changes that affect when servo power is enabled/disabled, so that might be the issue. An immediate workaround might be to flash a 2.5.x firmware version to your axidraw, but I'll plan to look into this and make it work with 2.6.x when I have time.

duncangeere, 2020-04-19:
Thanks! Appreciate you taking a look.

morozgrafix, 2020-04-22:
I'm having similar issue on Mac. Pen up/down don't work. After launching Inkscape with AxiDraw Control and moving pen up and pen down. I relaunch saxi for some reason hitting Pen Up button puts pen down, and Pen Down brings pen up. Starting the plot pen is always down and doesn't lift up at all. I'm on 2.6.5 firmware for Axidraw V3. Let me know if you need to run some logging or debug mode. Thanks.

dfeles, 2020-05-14:
Same here (im on raspberry pi 4), but I noticed, that when I print anything, after it stops, pen up/down starts working

p01c, 2020-10-20:
This problem can be fixed by sending 'SP,0', to the EBB, once it is conected:
http://evil-mad.github.io/EggBot/ebb.html#SP - This disables the power management of the lift servo.
I just created another button, sending the command however, would be nice to have it done on startup.

p01c, 2020-10-21:
If the servo is already sleeping, you can send the "TP" command to wake it up again.

[image]

gilgamec, 2020-10-24:
@p01c : That works! I dropped a `setPowerPlan` into `prePlot`, and saxi now actually picks up the pen!

oskay, 2020-11-11:
Likely will be resolved in a near-future firmware update: https://github.com/evil-mad/EggBot/issues/144

nornagon, 2020-11-11:
I think this is also now fixed in the `master` branch of saxi. I can't reproduce this issue on my own Axidraw as the hardware I have is the EBB v2.3 (on the Axidraw V2), which always powers the servo motors.

Would anyone who's able to reproduce this issue be able to try the following steps to run a pre-release version?

```
$ git clone https://github.com/nornagon/saxi
$ cd saxi
$ npm install
$ npm start
```

then visit `http://localhost:9080` and try a plot.

If anyone can confirm that the issue is fixed with the current `master` branch of saxi, I'll create a new release and mark this as resolved.

gilgamec, 2020-11-12:
I tried the new `master` branch (a6081c7). The "pen up" and "pen down" buttons still don't do anything. Starting a plot seems to work fine; the pen carriage moves up and down as expected during the plot. (Without being able to pre-stage, I couldn't get a pen in there.) No plot I've tried has gone more than a few seconds, though; they all eventually fail with errors like this one:
```
unexpected data: !5 Err: Need comma next, found: 'N'
(node:10756) UnhandledPromiseRejectionWarning: Error: Expected OK, got !5 Err: Need comma next, found: 'N'
    at EBB.<anonymous> (/home/pi/Source/saxi/dist/server/ebb.js:116:23)
    at Generator.next (<anonymous>)
    at RegexParser.<anonymous> (/home/pi/Source/saxi/dist/server/ebb.js:43:52)
    at RegexParser.emit (events.js:314:20)
    at addChunk (_stream_readable.js:307:12)
    at readableAddChunk (_stream_readable.js:282:9)
    at RegexParser.Readable.push (_stream_readable.js:221:10)
    at RegexParser.Transform.push (_stream_transform.js:166:32)
    at /home/pi/Source/saxi/node_modules/@serialport/parser-regex/lib/index.js:41:12
    at Array.forEach (<anonymous>)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:10756) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:10756) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
unexpected data: !5 Err: Need comma next, found: 'N'
```
I've never seen this error in my (crudely patched) d624691.

nornagon, 2020-11-12:
@gilgamec ah, i don't yet enable the servos when clicking penup/pendown. I think I've fixed the plotting issue in 4cc4f2d. Would you be able to give it another shot?

gilgamec, 2020-11-12:
It still crashes, with:
```
(node:12886) UnhandledPromiseRejectionWarning: Error: Expected OK, got !7 Err: Extra parmater
    at EBB.<anonymous> (/home/pi/Source/saxi/dist/server/ebb.js:116:23)
    at Generator.next (<anonymous>)
    at RegexParser.<anonymous> (/home/pi/Source/saxi/dist/server/ebb.js:43:52)
    at RegexParser.emit (events.js:314:20)
    at addChunk (_stream_readable.js:307:12)
    at readableAddChunk (_stream_readable.js:282:9)
    at RegexParser.Readable.push (_stream_readable.js:221:10)
    at RegexParser.Transform.push (_stream_transform.js:166:32)
    at /home/pi/Source/saxi/node_modules/@serialport/parser-regex/lib/index.js:41:12
    at Array.forEach (<anonymous>)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:12886) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:12886) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
unexpected data: OK
```

nornagon, 2020-11-12:
@gilgamec unfortunately I'm not sure which command is failing there. I've just pushed a new commit that adds extra debugging info when a command fails. Would you be able to try again with the latest code?

I've also added a `DEBUG_SAXI_COMMANDS` environment variable which will print out _all_ commands sent to the device. Hopefully it shouldn't be necessary to use that to determine the command that caused the error, but it's there if needed. To use it, run:

```
$ DEBUG_SAXI_COMMANDS=1 npm start
```

gilgamec, 2020-11-12:
It completed a bunch of times, but failed once:
```
unexpected data: !7 Err: Extra parmater
(node:13060) UnhandledPromiseRejectionWarning: Error: Expected OK, got !8 Err: Unknown command '!6:2136'
    at EBB.<anonymous> (/home/pi/Source/saxi/dist/server/ebb.js:116:23)
    at Generator.next (<anonymous>)
    at RegexParser.<anonymous> (/home/pi/Source/saxi/dist/server/ebb.js:43:52)
    at RegexParser.emit (events.js:314:20)
    at addChunk (_stream_readable.js:307:12)
    at readableAddChunk (_stream_readable.js:282:9)
    at RegexParser.Readable.push (_stream_readable.js:221:10)
    at RegexParser.Transform.push (_stream_transform.js:166:32)
    at /home/pi/Source/saxi/node_modules/@serialport/parser-regex/lib/index.js:41:12
    at Array.forEach (<anonymous>)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:13060) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:13060) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

nornagon, 2020-11-12:
hm, looks like the error-handling logic I added didn't kick in. Would you be able to try one more time with `DEBUG_SAXI_COMMANDS=1`? It should print out a whole lot of stuff when you run it (every command that's sent to the device).

Thanks so much for your help debugging this!

gilgamec, 2020-11-12:
I did get an error on my first attempt after plugging in the plotter, but everything I tried after that ran to completion (again, as far as I can tell since I can't stage the pen).
```
writing: EM,2,2
(node:13429) UnhandledPromiseRejectionWarning: Error: Error in response to command 'EM,2,2': Expected OK, got !8 Err: Unknown command '!1:2131'
    at EBB.<anonymous> (/home/pi/Source/saxi/dist/server/ebb.js:143:23)
    at Generator.throw (<anonymous>)
    at rejected (/home/pi/Source/saxi/dist/server/ebb.js:6:65)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:13429) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:13429) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
unexpected data: !7 Err: Extra parmater
```
Still, without the "pen up" and "pen down" buttons working, the fix is only halfway successful.

nornagon, 2020-11-12:
@gilgamec hm, sometimes there can be stale data left in the serial pipe that can cause issues when re-plugging the plotter after unplugging it without a full power cycle. saxi used to attempt to handle this (see https://github.com/nornagon/saxi/blob/75cff91b82ebee419046625335a99382606ee469/driver/src/saxi/EBB.scala#L328-L334, back when the project was written in Scala), but doesn't any more. I'd like to bring that logic back. For now, this kind of error is best addressed by unplugging the axidraw from all power sources, then replugging it in before using saxi.

The latest commit to master a7f55d0 should solve the issue with the pen up / pen down buttons.

gilgamec, 2020-11-12:
I still get a crash (with the same error) on the first plot, even after unplugging the AxiDraw. But after restarting saxi, everything after that seems to work fine. Thanks!

nornagon, 2020-11-13:
@gilgamec I just pushed c5e3c43 which attempts to mitigate the crash you're seeing. Would be great if you could test it out!

Regardless, I think this is definitely an improvement on the current latest release, so I'll spin a new release soon.

nornagon, 2020-11-14:
Released 0.12.0, which contains the fixes for this.

closed 2020-11-14 by nornagon

EmbeddedMan, 2020-11-23:
The underlying bug in the EBB firmware has been fixed as of the upcoming 2.7.0 release.