nornagon/saxi #48: Long loading and unreadable image with large svg

Makio64, 2020-04-20:
Saxi is slow to load large SVG and the image cant be preview correctly due to the path over it.

[image]

joining nearby paths: 117294.57983398438ms
background-planner.ts:76 planning pen motions: 1688.39501953125ms
background-planner.ts:11 serializing: 657.703125ms
ui.tsx:229 deserializing: 4153.98486328125ms

I used a grid strategy to reduce the number of collision test on my polygons/lines visibility, and it improves by almost 100 the time to generate my isometric city, I think similar algorithm might help here.

About complex SVG getting impossible to read due to the path, an option to hide the plotter path would be an easy and quick fix.

nornagon, 2020-04-21:
I'm surprised that saxi would path across the image that heavily! Would you be able to attach the SVG so that I can try out some performance tweaks?

`joinNearby` is O(n), it doesn't do any spatial searches. https://github.com/nornagon/saxi/blob/05b0b7b645d2195928f090cbbd42a4f0946b9f22/src/optimization.ts#L33

Makio64, 2020-04-21:
I usually don't sort the path sorting cause I want to feel more the algorithms while it's drawing

cant attach SVG but here is two links:
[](https://makioandfloz.com/tmp/test_medium.svg )( 6mo )
[](https://makioandfloz.com/tmp/test_hardcore.svg) ( 22 mo )

Thanks!

Makio64, 2020-05-11:
Hi, @nornagon did you have the opportunity to test them?