nornagon/saxi #49: Export optimized SVG

Jip-Hop, 2020-06-24:
Hi nornagon,

How would you feel like an "Export SVG" feature? I find saxi the most sexy and fast option to optimize  an SVG for plotting. I have a Makelangelo machine, not an Axidraw, but I'd still use saxi to optimize my files before plotting if it had an export button. I quickly tested the saxi optimized output, by copy pasting the contents of the SVG tag in the DOM, but it includes extra stuff like the "pen up" moves.

The fact that saxi also reverses paths when optimizing is super. The official Axidraw Inkscape plugins don't do that. And saxi doesn't need Inkscape, which is another big plus for me.

The only other program I found which optimizes SVG with similar efficiency is LightBurn. It exports to gcode, which would work for me as well. But I still like saxi better.

Will you consider adding this feature?

Jip-Hop, 2020-06-26:
This snippet will let you download the optimized SVG with the pen-up moves stripped out.
```javascript
var svgData = document.querySelector(".preview > svg").outerHTML;
// Get rid of the pen-up lines, based on their color
svgData = svgData.replace(/<\/path>/g, "</path>\n").replace(/<path.*rgba\(0, 0, 0, 0.3\).*<\/path>/g, "").replace(/\n/g, "");
var svgBlob = new Blob([svgData], {type:"image/svg+xml;charset=utf-8"});
var svgUrl = URL.createObjectURL(svgBlob);
var downloadLink = document.createElement("a");
downloadLink.href = svgUrl;
downloadLink.download = "saxi-optimized.svg";
document.body.appendChild(downloadLink);
downloadLink.click();
document.body.removeChild(downloadLink);
```
Can be copy pasted into the JavaScript console. But I still think an export button would be nice. 🙂

nornagon, 2020-06-26:
Seems like a reasonable thing to add.