nornagon/saxi #52: Support for clipPath

jvolker, 2020-07-30:
At the moment clipPath is not used. Would be great to get support for it.

Original file in Affinity Designer:
[image]

In saxi:
[image]

<details><summary>Source of SVG</summary>

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="100%" height="100%" viewBox="0 0 596 596" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" xmlns:serif="http://www.serif.com/" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linecap:round;stroke-linejoin:round;stroke-miterlimit:1.5;">
    <rect id="Artboard1" x="0" y="0" width="595.276" height="595.276" style="fill:none;"/>
    <g id="Artboard11" serif:id="Artboard1">
        <g transform="matrix(1,0,0,1,-4.28127,-40.7584)">
            <clipPath id="_clip1">
                <circle cx="301.919" cy="338.396" r="232.125"/>
            </clipPath>
            <g clip-path="url(#_clip1)">
                <g transform="matrix(1,0,0,0.975887,0,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,20.1848,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,40.3696,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,60.5543,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,80.7391,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,100.924,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,121.109,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,141.293,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,161.478,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,201.848,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,222.033,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,302.772,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,181.663,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,242.217,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,282.587,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,343.141,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,383.511,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,423.88,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,262.402,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,322.956,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,363.326,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,403.696,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,444.065,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
                <g transform="matrix(1,0,0,0.975887,464.25,2.52033)">
                    <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:1.01px;"/>
                </g>
            </g>
        </g>
    </g>
</svg>
```
</details>

I've also tried several methods and other software (Illustrator, Inkscape) to convert this clipPath/mask into lines. Without luck. I'm happy about any suggestions.

Thanks.

nornagon, 2020-07-30:
This is a nontrivial amount of computational geometry, unfortunately. The work would have to happen in [flatten-svg](/nornagon/flatten-svg).

msurguy, 2020-07-30:
@jvolker I think it's worth pointing out that ClipPath and masks are virtual features of SVG, which do not translate into CNC operations in a meaningful way because CNC machines work by moving along vector paths. ClipPath is only virtually hiding the paths / vertices and not really changing the SVG paths, so the machine cannot distinguish between clipPath and no clipPath.

In order to get the output desired, you'd need to actually modify the paths (trim, crop). I was working on a tool that can do that for you but never finished it. 

You can check it out here: https://msurguy.github.io/svg-cropper-tool/
- You'd need to have your SVG document size explicitly defined (width / height)
- You need to have no matrix transforms in the document (I use https://jakearchibald.github.io/svgomg/ to simplify / convert paths, sometimes need to open Illustrator for final tuning and export)

jvolker, 2020-07-31:
Thanks for your fast replies.

@nornagon feel free to [move this issue](https://docs.github.com/en/github/managing-your-work-on-github/transferring-an-issue-to-another-repository) to the flatten-svg repository.

@msurguy Thanks for explaining this in more detail. I am aware of the problem but am surprised there aren't more tools implementing this yet.

Thanks for mentioning your tool. When you are saying you've never finished it, what's still missing?

I did a little more research on this topic and found the following code fragments/projects that might be useful:

https://github.com/ubilabs/axidraw/blob/master/src/lib/crop-lines-by-bbox.js
https://github.com/ubilabs/axidraw/blob/master/src/lib/crop-lines-by-circle.js

And these two:
https://github.com/w8r/martinez/ (mentioned in https://github.com/svg/svgo/issues/680)
https://github.com/velipso/polybooljs (used in https://github.com/w8r/martinez/issues/123)

jvolker, 2020-07-31:
Two more: 

https://github.com/mfogel/polygon-clipping
https://github.com/johnrees/clipper-fpoint

msurguy, 2020-07-31:
@jvolker thank you, I'm fully aware of those algos + quite a few more since I had to implement clipping into my tool to be able to crop shapes. Did you try the tool? 

What's not done in my tool (but could be done if somebody can contribute):
- Smarter evaluation if each path even needs to be cropped (currently there's no check, and ClipperJS sometimes deforms the path a tiny bit)
- The SVG doc needs to be first optimized (all shapes converted to paths, remove clipPaths, unsupported features, etc). I have been able to run SVGO in the browser, and that tool can do all of that, just needs to be integrated into this product. Here's my browser based SVGO: https://codesandbox.io/s/svgo-browser-03erd?file=/index.html
- Ability to load arbitrary SVG path to be used as a cutout border
- Restoring path coloring / width / fills for the paths that were originally colored.

If you tried the tool - let me know what you think!

msurguy, 2020-07-31:
PS, @nornagon I'm using Flatten-SVG in the cropper tool :) thank you so much for it!

jvolker, 2020-08-08:
I've given some feedback in the other thread. Your tool looks really promising. Thanks for that.

## Maybe this is helpful to some people in the meantime:
I've realized manually removing vectors was far easier and closer to what I wanted using Illustrator compared Affinity. The "Lasso tool", "Erasor tool" and of course "Direct Selection Tool" of Illustrator can be helpful.

Also **Affinity** doesn't cut the paths:
[image]

**Illustrator** cuts the paths (closer to what I wanted to achieve):
[image]