nornagon/saxi #53: Plotter bumps into its boundaries

jvolker, 2020-07-30:
I've had the plotter bumping into its frame when trying to draw outside the drawing boundaries. I loaded an SVG with a clipPath and disabling "fit page".

The following is a file that reproduces this.

In Affinity Designer:
[image]

In saxi:
[image]

<details><summary>Source of SVG</summary>

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="100%" height="100%" viewBox="0 0 596 596" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" xmlns:serif="http://www.serif.com/" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linecap:round;stroke-linejoin:round;stroke-miterlimit:1.5;">
    <rect id="Artboard1" x="0" y="0" width="595.276" height="595.276" style="fill:none;"/>
    <clipPath id="_clip1">
        <rect id="Artboard11" serif:id="Artboard1" x="0" y="0" width="595.276" height="595.276"/>
    </clipPath>
    <g clip-path="url(#_clip1)">
        <g transform="matrix(1.27366,1.27366,-1.27366,1.27366,342.983,-517.669)">
            <g transform="matrix(1,0,0,0.975887,0,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,20.1848,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,40.3696,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,60.5543,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,80.7391,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,100.924,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,121.109,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,141.293,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,161.478,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,201.848,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,222.033,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,302.772,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,181.663,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,242.217,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,282.587,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,343.141,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,383.511,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,423.88,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,262.402,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,322.956,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,363.326,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,403.696,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,444.065,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
            <g transform="matrix(1,0,0,0.975887,464.25,2.52033)">
                <path d="M69.794,104.522L69.794,582.035" style="fill:none;stroke:black;stroke-width:0.56px;"/>
            </g>
        </g>
    </g>
</svg>

```
</details>

Thanks.

nornagon, 2020-07-30:
This is expected behavior. Saxi doesn't clip the paths to the page area. If you disable 'fit to page' it's up to you to make sure your lines stay within the page :)

nornagon, 2020-07-30:
(Though I could imagine adding an option to clip to page bounds! That might be useful.)

jvolker, 2020-07-31:
>  adding an option to clip to page bounds! That might be useful.

Yes, I think it would be really good also to prevent damage to the machine. And, when this bumping happens the machine loses track of its position which requires to start the plot from scratch. So would be good to avoid that.

This could help to implement it:
https://github.com/ubilabs/axidraw/blob/master/src/lib/crop-lines-by-bbox.js

nornagon, 2020-08-15:
That function you linked would remove every line in your example, since all of them have a start or end point outside the bounding box. So we'd need a more complex algorithm, e.g. Liang-Barsky https://gist.github.com/ChickenProp/3194723.

jvolker, 2020-08-15:
Or could implementing parts of https://github.com/msurguy/svg-cropper-tool help with this too?

nornagon, 2020-08-15:
@jvolker If you're able to try master, give it a shot. I've implemented a cropping algorithm. I haven't tried it myself yet (though I did write some tests). I'll wait to make a release until I'm able to test it a bit more thoroughly, but if you're game to try out the bleeding edge, it should work!

closed 2020-08-15 by nornagon

jvolker, 2020-08-16:
Beautiful. I've just tried with the above SVG and it works perfectly. Thanks.

commit a1f0bfe1 (2020-08-15): feat: crop to margins when not fitting to page

Fixes #53
commit 1bf1d302 (2020-09-16): feat: crop to margins when not fitting to page

Fixes #53