nornagon/saxi #54: Slightly stuttering with Raspberry Pi Zero

jvolker, 2020-08-07:
I've tried to use this with a Raspberry Pi Zero. In comparison to using it on a Mac Book Pro (2,5 GHz Intel Core i7, 16GB RAM) the movement of the plotter (AxiDraw V2) a little less smooth. It's okay for the most part, but every now and then the plotter stutters for a fraction of a second.

- Could the Pi zero be to slow for this job? Could using a Pi 3 solve this?
- Could it be an issue with the wifi (which seems quite stable)?

Thanks.

nornagon, 2020-08-15:
The wifi shouldn't be an issue—all the information needed for the plot is sent to the server before the plot is started. My best guess is that JavaScript's garbage collection kicking in is the reason for the stutter.

closed 2020-08-15 by nornagon

jvolker, 2020-08-15:
I've switched to Pi 3B+ and it's now working like a charm.