nornagon/saxi #55: Saxi stopped during long print

jvolker, 2020-08-07:
The plotter stopped close to finishing this print:
[image]
[image]
[taranaki-rivers-bug.svg.zip](https://github.com/nornagon/saxi/files/5042086/taranaki-rivers-bug.svg.zip)

The UI was unresponsive (tried to press pause/resume which didn't work). On the board the USB led was flashing and DIR1 led was on. I had to restart saxi. Even then the motors were still engaged initially. After the restart of saxi it seemed to behave normally.

**OS**
macOS 10.14.6

**Saxi version**
0.9.2

**AxiDraw V2**
EBBv13_and_above EB Firmware Version 2.6.5

Might be related to https://github.com/nornagon/saxi/issues/31.

nornagon, 2020-08-15:
That is concerning :( I'm not sure what the USB and DIR1 leds mean on the EBB... if it worked for you when you tried again, then the next step on tracking this down is probably to add a bunch of logging information to saxi so we can see exactly where it gets stuck.

jvolker, 2020-08-15:
I've not yet tried this another time yet but could do that in the next few days.

joseedil, 2022-08-31:
@jvolker do you remember if the plotter started to get sluggish before stopping?

Yesterday I was doing a long plot with saxi on a Raspberry Pi V3 1GB RAM and the same happened.
I was using this SVG, which has around 900 paths, each with 12 lines and some "move to" commands.

[image]

I was able to make a successful plot of a smaller version of this, with 352 paths. The machine was sluggish at first, but after a few minutes, it went smoothly.

I monitored memory usage on a second try. It starts to rapidly increase memory usage and get sluggish (the plotter). It halts when the swap space ends until the OS finally kills the process for using all memory in the machine. Saxi works after restarting the process, but there's no way to resume the plot.

I'm not familiar with JS or TS, but I'm getting myself into the code base to understand how it works and try to help improve it. Seems something related to trying to build and keep the whole sequence of commands in memory.

@nornagon are you still maintaining saxi and accepting pull requests?

joseedil, 2022-08-31:
Ah, I'm using saxi from the master branch compiled on the machine as I needed to change the [stepsPerMm](https://github.com/nornagon/saxi/blob/f31840fd5747de8b5fe1c215872abf44c1aaf97b/src/planning.ts#L92) parameter to `1.25 * 5` to match my machine (iDraw A3 v1). I'm running EBB firmware v2.8.1 (latest)

nornagon, 2022-08-31:
Hi @joseedil, yep I'm still around and contributions are welcome! It makes sense that saxi would start to slow down and eventually crash in a low-memory environment; I haven't extensively tested or optimized for that. It ought to be possible to make some architectural changes such that if a plot is successfully planned then it will be guaranteed to execute to completion without running out of memory.

(are you speedrunning the [250 box challenge](https://drawabox.com/lesson/250boxes)? 🤣)

joseedil, 2022-08-31:
Not really, I went through Draw a Box challenge by hand a few months ago 😅, but it is surely an inspiration for this.

I tried the Axidraw CLI in the meanwhile and it handles this SVG on the Pi using around 40MB of memory (but it doesn't have the nice web interface of saxi).