nornagon/saxi #56: Raspberry Pi install instructions

jvolker, 2020-08-14:
On Raspberry Pi 3B+ with latest Raspbian Buster.

`npm i -g saxi` gave me an error: `npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules`
I tried `sudo npm i -g saxi` and got: 
```
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/local/lib/node_modules/saxi/node_modules/@serialport/bindings/.node-gyp'
```

`sudo npm i -g saxi --unsafe-perm` worked for me.

nornagon, 2020-08-15:
This is a setup issue on your Raspberry Pi. Installing npm modules with `sudo` is not recommended.

closed 2020-08-15 by nornagon

jvolker, 2020-08-15:
In case it's useful to someone: 
I used this guide: https://www.instructables.com/id/Install-Nodejs-and-Npm-on-Raspberry-Pi/ which came up first in a Google search. Looking at it again it seems like you can also use `apt-get` to install Node js. Maybe this makes the difference.