nornagon/saxi #59: Up and down height direction is counterintuitive

jvolker, 2020-08-16:
I think the up and down height direction is counterintuitive. The up position should have a higher number compared to the down position. Increasing the numbers should lift the pen up, not down.

Would be great to invert those numbers. Thanks.

nornagon, 2020-08-17:
I disagree :)

jvolker, 2020-08-17:
Why?

jvolker, 2020-08-17:
Is there a reason? Did you implement this from a technical perspective, do you think it's more intuitive for an average user or is this simply your personal preference?

I'm a user who has recently started using plotters.

Thanks.

nornagon, 2020-08-17:
This matches the way the firmware works.

I think about it as _how "down" is the pen, on a scale from 0% to 100%_.

Now that I look, it seems like the Inkscape driver maybe has things the other way around, which is a reasonable reason to think about switching it: https://github.com/evil-mad/axidraw/blob/74b7e800c70a9284d6ff7e182b21603c81c7e945/inkscape%20driver/axidraw_conf.py#L39-L40

EmmaSimon, 2021-06-22:
Just another perspective, but I was confused about this at first after being familiar with Inkscape.

ianstormtaylor, 2022-05-09:
Yeah was also confused by this on first use, and have to re-orient whenever I need to adjust the pen height again (which luckily isn't too often).

Either way, thank you for Saxi!

alexrudd2, 2023-07-21:
+1 on this. I thought it was a bug until finding this explanation. :)