nornagon/saxi #60: Payload too large error

Jip-Hop, 2020-09-08:
I'm trying to draw some heavy SVG files. I kept getting a `PayloadTooLargeError: request entity too large` error.
It can be fixed by choosing a value higher than 100mb here:
https://github.com/nornagon/saxi/blob/8c807ad971f7ead5946506ad5c538398b3f5232c/src/server.ts#L16
I changed this value to 1000mb myself in this file after globally installing saxi:
`/node_modules/saxi/dist/server/server.js`

nornagon, 2020-09-08:
Thanks for the report! I wonder if there's a way we can stream the upload so we can do unlimited size...

nornagon, 2020-09-22:
Should be fixed by c226d75.

closed 2020-09-22 by nornagon

Jip-Hop, 2020-09-22:
Thanks for the fixes :) I can confirm it now works with my test svg file.

However, when I replace the squares in the test svg with a more complex shape (e.g. a star) the number of anchor points increases and the file gets much bigger. I've got a couple of 60MB svg files I'm trying to plot. After importing the SVG, the "Replanning" phase starts. But it eventually crashes the page with an "Out of Memory" error. The PC has enough RAM and it's nowhere near it's limit though.

nornagon, 2020-09-22:
Try this:
https://stackoverflow.com/questions/17491022/max-memory-usage-of-a-chrome-process-tab-how-do-i-increase-it

j

Jip-Hop, 2020-09-26:
Thanks for replying :) I tried that, but can't tell if it helped. The page still crashes and gives an Out of Memory error. Sort paths is off, and the point-joining radius and path-joining radius are 0, so I'm a bit surprised it needs so much memory.