nornagon/saxi #62: Page crash when starting long plot

Jip-Hop, 2020-09-21:
[image]

The page crashes when trying to start a long plot. Doesn't matter if the Axidraw is connected or not. Tried on both Windows and MacOS. The file itself isn't the problem, I tried many different files. Also, the same file which causes a crash will start to work when increasing the "path-joining radius" setting until the draw time is down to about 9 hours.

This is what it looks like before I press the "Plot button":
[image]

After pressing "Plot" the page crashes and the drawing isn't started. Saxi itself doesn't crash and I can reload the page to start over...

I've got no clue how to debug this. Rather nasty bug. But I've included a file which you can use to replicate the issue:

[crash_test.svg.zip](https://github.com/nornagon/saxi/files/5255382/crash_test.svg.zip)

nornagon, 2020-09-22:
Yep, I can reproduce it. This seems to be a Chrome bug: https://crbug.com/835077. It happens when you try to send a `POST` request that's larger than 128 MB. So this code for instance triggers the crash:

```js
fetch("/", { method: "POST", body: new Uint8Array(1024*1024*256) })
```

It looks like wrapping this in a `Blob` will work around the issue (thanks [stack overflow](https://stackoverflow.com/questions/51882961/increase-maximum-body-size-for-the-fetch-api-in-chrome)), but it's still likely a bit shaky; the server currently has a 100 MB limit on uploads. That can be expanded—this particular file is only slightly larger, at 136 MB. Ideally we'd use something like [streaming requests](https://web.dev/fetch-upload-streaming/), but that's not yet generally available.

I'll wrap the request in a Blob and bump the upload limit up a bit (and make it configurable), which should solve this issue.

nornagon, 2020-09-22:
Fix released in 0.11.0.

commit c226d754 (2020-09-22): fix: work around a Chrome crash with large payloads

Fixes #62.
closed 2020-09-22 by nornagon