nornagon/saxi #63: Drive Axidraw over WebSerial

nornagon, 2020-09-22:
Using the [Web Serial API](https://web.dev/serial/), currently in origin trial at time of writing.

matthewjacobson, 2021-07-14:
Hello! Curious if you were able to make any progress on this?

I have been working on building out my own library but kind of stuck at the moment (https://github.com/matthewjacobson/axidraw.js) - it is kind of working but having weird random issues.

I would be happy to help out here if I can!

nornagon, 2021-07-14:
@matthewjacobson yep, I spent some time on this last week, there's actually a prototype of it in the repo currently. You can try it out at https://nornagon.github.io/saxi/. Still some issues to work out around reconnection when the device is disconnected, but I've successfully plotted stuff from it!

matthewjacobson, 2021-07-14:
@nornagon oh awesome!! Is it still running over a server or is it running locally in the browser?

nornagon, 2021-07-14:
WebSerial is local in the browser.

matthewjacobson, 2021-07-14:
Amazing! Happy to help out with anything if you need a hand

nornagon, 2021-07-14:
@matthewjacobson Just realised I hadn't pushed up my prototype code. Did that now.

The issue i'm currently a bit perplexed by is that it seems impossible to close a WebSerial port after opening it. To repro in saxi currently, switch `!driver` to `true` here: https://github.com/nornagon/saxi/blob/54e6f55615fb1e919f787cb2c3301b4951cda2b7/src/ui.tsx#L988, then click 'Connect', then click 'Connect' again.

closed 2021-07-14 by nornagon

matthewjacobson, 2021-07-14:
Interesting - I think I have disconnect working in my version (https://github.com/matthewjacobson/axidraw.js/blob/ff6908057ad4cc1533e7d0a533469fd71b849d93/src/axidraw.js#L69).

You can test it here https://iammattjacobson.com/axidraw.js/index.html

But maybe I am missing something - I was mostly following the few examples I found online.

nornagon, 2021-07-14:
ah hm... i'm piping to a WritableStream which is maybe the issue: https://github.com/nornagon/saxi/blob/54e6f55615fb1e919f787cb2c3301b4951cda2b7/src/ebb.ts#L29-L31

matthewjacobson, 2021-07-16:
@nornagon would you mind sharing the command you use to build/compile the project?

nornagon, 2021-07-16:
It's all in package.json. npm start or npm run dev :)

matthewjacobson, 2021-07-16:
Ah I see - I had to add "IS_WEB=1" to the run script. But looks like it is working well now. Thanks!

mutoo, 2021-08-01:
This is so cool! I am also working on a prototype that communicates Axidraw over the WebUSB on Chrome. It's work-in-progress but the communication part is pretty stable at the moment. Hope I can work it out public on GitHub soon.

nornagon, 2021-08-03:
@mutoo Interesting! What's the goal of using WebUSB as opposed to WebSerial?

mutoo, 2021-08-03:
@matthewjacobson @nornagon I had made the repo public https://github.com/mutoo/axidraw-web so that everyone can play with it (it's not that powerful at the moment).

Actually, I am not sure what's the difference between WebUSB and WebSerial. The initial idea of making that project is building a web app that can control AxiDraw without setting up a node js environment. And WebUSB first came into my mind.

The only thing different I found is that I have to set up SSL for the browser to communicate with WebUSB, but I don't see a similar setup in saxi.

nornagon, 2021-08-04:
@mutoo saxi's webserial version works over https, that's a browser restriction. https://nornagon.github.io/saxi/ is the UI :)

commit 593c451e (2021-07-14): feat: WebSerial support

Closes #63.