nornagon/saxi #65: Axidraw reaching beyond limits

mmattyg, 2020-10-18:
Drawing with A4 paper, 25 mm margin, Older Axidraw model (V2?) with Firmware 2.6.5

The axidraw is trying to reach over  its limits (pulls too much back or too far away) and grinds the gears.
What am I doing wrong?

Thanks!

mmattyg, 2020-10-18:
Possibly paper should always be kept in landscape?

nornagon, 2021-07-04:
@mmattyg the Axidraw V2's work envelope is 300 × 218 mm, and an A4 sheet is 297 mm on its long side, so you must orient the paper in landscape if you want the Axidraw to be able to reach it all.

Unfortunately I'm not sure the EBB firmware offers any support for querying work area limits. I'm not sure how the Inkscape driver achieves it, but without some way to know the size of the work area, saxi can't implement this in an automated fashion. Either EBB will have to add support for querying work area limits (which I think would be technically challenging to retrofit), or the user would have to manually enter the work area limit, which I don't think is a great experience :/

jonsimington, 2024-02-04:
I am also experiencing this issue.  

saxi 0.15.0 running on a RaspberryPi 4, EBB 2.8.1 using an AxiDraw V3/A3.  

I have my paper size set to ArchA, and the Axidraw is attempting to draw off the paper (I have my paper aligned properly to the origin), and sometimes it seems to attempt to move the pen further right than my AxiDraw physically allows, causing some nasty noises.

What can I try next?  It seems like folks have success with saxi on raspberry pi, but my trials are unsuccessful, and I'm scared to keep testing for fear of mechanically breaking my AxiDraw.