nornagon/saxi #69: Geometry-based settings

HenrySkup, 2020-12-01:
I was wondering if it would be possible to "pull out" machine settings on a geometry / layer level.  Something similar to [AxiDraw's Layer Control](https://wiki.evilmadscientist.com/AxiDraw_Layer_Control).  If the interface is predictable, then it could be some pretty powerful functionality.

Use cases: 
you are creating an SVG of hash lines each line you want to set to a different pen speed (would get different ink bleeding)

nornagon, 2020-12-01:
To clarify—are you suggesting specifying the machine settings in the SVG? Perhaps something like this?

```svg
<path saxi:speed="100" d="..." />
<path saxi:speed="150" d="..." />
<!-- ... -->
```

HenrySkup, 2020-12-02:
**yes, something like that for sure.** 👍 

Sorry for the long read, but just wanted to get some thoughts down:

seems like Layer Control allows for control over : 

1. 'pen-down-speed'
2. 'pen-down-height'
3. 'delay-before-pen-down'
4. 'do-not-plot'
5. 'pause-for-button-press'

This could be expanded, of course, these things come to mind :
- other settings which might be useful:
    - 'delay-after-pen-down'
    - 'delay-before-pen-up'
    - 'delay-after-pen-up'
    - 'pen-down-height'
    - 'pen-up-height'
    - 'curve-resolution'
    - 'pen-up-speed'
    - 'pen-lower-speed'
    - 'pen-raise-speed'
- change plotter state (`nodecount`, other settings) 
- you could imagine triggering all sorts of actions/settings at certain geometries or (even vertex points along paths)
    - UI prompts (perhaps message with the 'pause-for-button-press')
    - make EBB commands (activate other pins)

----

Two things about the *Layer Control implementation* which seem sub-optimal :

#### Applicable Elements

It only works fo <g> and not geometry elements -- your suggestion above (along with the ability to mark up svg container elements) is what i was thinking

#### Inheritance

The "inheritance" model doesn't seem supper intuitive.  Right now there is no inheritance, i would think that something more DOM/CSS like would make more sense for people. 

##### My suggestion
There would be "root" level settings that could be overridden at a container or geometry level.  *Some* settings would be inherited, others (ex. 'pause-for-button-press') would not.

It could be interesting to **also** have the option to do something like 

```
<saxi:settings>
    <saxt:preset name="fast" saxi:speed="200" ... /> 
    <saxt:rule for="circle" saxi:resolution="0.8" ... /> 
</saxi:settings>

<!-- ... -->
<path saxi:preset="fast" d="..." />
<path saxi:speed="150" d="..." />
<circle ... />
<!-- ... -->
```

or even 

```
<saxi:settings>
    .fast {
        speed : 200;
    }
    circle {
        resolution : 0.8;
    }
    .slow circle {
        resolution : 0.9;
    }
</saxi:settings>

<!-- ... -->
<path class="fast" d="..." />
<path saxi:speed="150" d="..." />
<circle ... />
<g class="slow">
    <circle ... />
</g>
<!-- ... -->

```

I don't know if this is "reinventing" CSS, but it would allow for specifying settings across the hierarchical inheritance.  Though I can imagine that it could complicate things?