nornagon/saxi #70: 0 radius circles only work when "fit page" selected

HenrySkup, 2020-12-01:
Hi, I'm trying to plot points (pen down + delay + pen up ).  Saxi seems to be able to at least do the pen down + pen up motion when given `<circle>`s with `r=0`.  However it is only generating a plan when the "fit page" option is selected.  When this is deselected, the preview disappear (i.e. no travel paths) and nothing plots.

Stanko, 2020-12-01:
I wasn't aware you could plot circles with `r=0`. What I usually do is to generate a super short path, something like:
```js
`<path d="M ${ p.x } ${ p.y } H 0.01" />`
````
which draws a horizontal line, but it is short enough to plotted as a dot. 

This works with saxi and with the official inkscape plugin. Hope that helps!

nornagon, 2020-12-01:
@HenrySkup would you be able to attach an SVG that exhibits the issue you're encountering?

HenrySkup, 2020-12-02:
@nornagon 

Here are the SVGs :
[Archive.zip](https://github.com/nornagon/saxi/files/5627681/Archive.zip)

**Notes**
Made with SVG.js, saved via [d3-save-svg](https://github.com/edeno/d3-save-svg)

HenrySkup, 2021-11-19:
I think this has been resolved.  Just did a plot yesterday with the latest version and was able to plot a grid of "dots"

closed 2021-11-19 by HenrySkup

nornagon, 2021-11-22:
Glad to hear it!