nornagon/saxi #82: Help setting up new servo

jimmykl, 2021-04-21:
I have installed a servo which reverses the axidraw's up/down controls

Plots work ok by swapping the up/down % in saxi's ui but when the plot finishes the final move is done with the pen down and I get a line drawing back to the home position over the plot 😢 

I've got a dev build running but can't follow the code well enough to see where the pen height during the final move is controlled

I thought it could be in `server.ts` and tried adding `ebb.setPenHeight(Device.Axidraw.penPctToPos(100), 1000);` in a few places (`realPlotter` and the `doPlot` function) but to no effect and I'm really just shooting blind

It would be great if someone could point me in the right direction!

jimmykl, 2021-04-22:
Found it (in `planning.ts` `plan`)

closed 2021-04-22 by jimmykl

nornagon, 2021-04-27:
Oh dear! I'm glad you figured it out. Do you think it would be broadly useful to have a checkbox to reverse servo direction in the options? Or is this particular to your individual setup?

jimmykl, 2021-04-27:
I think it's probably too much of an edge case to expose in the UI.

Unless there was an 'advanced: only change these if you're sure you need to' type of section?

nornagon, 2021-04-27:
Alrighty, let's leave it then :)