nornagon/saxi #86: No EBB Connected error for iDraw

EmmaSimon, 2021-06-21:
Saxi doesn't recognize the EBB of an [iDraw plotter](https://uunatek.com/product/idraw-handwriting-drawing-machine/). It still uses EBB, and is flashed with firmware v2.7.0. It works with Inkscape's Axidraw controls, not sure if the issue is the different hardware somehow. Running `saxi --firmware-version` just gives me `No EBB connected`.

EmmaSimon, 2021-06-21:
Huh, it connects and plots if I use `sudo` 🤔 This doesn't seem ideal, but it might be a permissions issue. I'll try digging more later.

nornagon, 2021-06-21:
Thanks for the report! If it works with `sudo` then that definitely seems like it could be permissions-related. Let me know what you find!

EmmaSimon, 2021-06-22:
Ok, haven't had a chance to look into the permissions yet, but I found something else. Not sure if this is related to the plotter or what, or if I should open a different ticket for it, but every time a plot finishes, instead of raising the pen and returning home, it instead presses the pen down much harder into the paper than it's configured to, and returns to home. You can see this in the attached images, in the first one it's the line through the 2deg and 1mm circles. It didn't leave a mark because it pressed hard enough to bend the nib sideways. In the other one, it left a deep gouge in the paper, it was pressing significantly harder than the setting for the regular strokes.

[image][image]

jimmykl, 2021-06-22:
That's what my axidraw was doing when I changed the servo and reversed the up/down heights.

I did this to get it working https://github.com/jimmykl/saxi/commit/81c7872fb0ab89e016ee9c7b009e7c081e069c62 You might need to tweak the heights to get it working on your machine.

@nornagon maybe it would make sense to add a config somewhere to set the 'final move height' or 'post plot height' to help troubleshoot for different plotters/servos?

EmmaSimon, 2021-06-22:
Why does the ternary on line 572 have to exist at all? Why not just return to the regular up position when the plot is done?

nornagon, 2021-06-22:
@EmmaSimon it's that way because in some cases when the paper is uneven, the normal 'up' height can end up leaving the pen tip still touching the paper. I wrote it to go to full-up height at the end of the plot so it would avoid dragging the tip across the paper at the end of the plot even if I'd gotten the up-height a little bit wrong. It looks like it backfired in your case though!

This seems unrelated to the top post though, so please open a new issue for further discussion :)

nornagon, 2021-07-04:
Closing this as it seems to be a setup issue, not a bug with saxi.

closed 2021-07-04 by nornagon

EmmaSimon, 2021-07-04:
Update for anyone else running into something similar: my npm global install location was `/usr` so global installs only worked as root and that was probably messing up the permissions, so I switched to a directory in home to install things globally without sudo. I also added my user to the `tty`, `dialout`, and `netdev` groups. Now saxi recognizes the EBB, but only if I specify the device with `-d` 🤷  Might try to figure out why it doesn't recognize it automatically at some point (when I looked at the code before, the [check for vendor and product id](https://github.com/nornagon/saxi/blob/main/src/ebb.ts#L14) should be true) but for now this is fine.