nornagon/saxi #87: Make pen height at the end of a plot configurable

EmmaSimon, 2021-06-22:
On my iDraw plotter, every time a plot finishes, instead of raising the pen and returning home, it instead presses the pen down much harder into the paper than it's configured to, and returns to home.

This seems to be the result of a reversed pen servo, and a fix for it was made on @jimmykl's fork here: https://github.com/jimmykl/saxi/commit/81c7872fb0ab89e016ee9c7b009e7c081e069c62

It might make sense to make the final up height configurable for general issues with different plotters and servos, rather than hard coding it to 0.

You can see this in the attached images, in the first one it's the line through the 2deg and 1mm circles. It didn't leave a mark because it pressed hard enough to bend the nib sideways. In the other one, it left a deep gouge in the paper, it was pressing significantly harder than the setting for the regular strokes.

[image][image]

_Originally posted by @EmmaSimon in https://github.com/nornagon/saxi/issues/86#issuecomment-865497479_

nornagon, 2021-06-22:
Would it make sense to check the "polarity" of the up/down heights that are already configured in the UI, and go to either 0% or 100% depending on which way "up" is?

i.e. if up < down, go to 0%, otherwise go to 100%?

EmmaSimon, 2021-06-22:
Ahh, yeah I like that, that should work well! Definitely would prefer to avoid a config if possible.

nornagon, 2021-07-04:
@EmmaSimon give 0.14.0 a whirl!

closed 2021-07-04 by nornagon

jimmykl, 2021-07-04:
@nornagon this works for me 🙏

EmmaSimon, 2021-07-04:
Yep, that works! Thanks so much 🎉

nornagon, 2021-07-07:
FYI, I opened https://github.com/nornagon/saxi/issues/90 to track some additional changes that are needed to fully support this. So be aware, you might run into issues when pausing/canceling, or altering pen heights in a plan.

commit c6bd5a7e (2021-07-04): fix: invert last pen-motion when up/down positions are inverted

Closes #87