nornagon/saxi #90: Various fixes needed to handle inverted up/down sense

nornagon, 2021-07-07:
(see #87 for context on when up/down sense is reversed)

### When pausing
https://github.com/nornagon/saxi/blob/117161d27d9b37af8aae71dbc36d123cb92a96bf/src/server.ts#L198

### When canceling
https://github.com/nornagon/saxi/blob/4d5c9918deb6f79bf97ac8eb561b39c7d668790e/src/server.ts#L161

### When altering pen height of a plan without replanning
https://github.com/nornagon/saxi/blob/c6bd5a7ef984d9ca537904e6d58ce7490ca8a2e7/src/planning.ts#L291-L298

---

All of these need to be updated to take into account when the pen up position and pen down position are reversed, similar to:

https://github.com/nornagon/saxi/blob/c6bd5a7ef984d9ca537904e6d58ce7490ca8a2e7/src/planning.ts#L566

EmmaSimon, 2021-07-08:
I noticed this last night and was thinking about putting together a PR. Have you started on the changes already?

nornagon, 2021-07-08:
Not yet! Happy to have contributions!

EmmaSimon, 2021-07-08:
Been looking at the code and I'm not sure I have time to figure this out in the next few days if you think you can throw something together sooner. The change for altering height in planning.ts is super straightforward, but for pause/cancel I'm unsure about the best way to get the pen up/down configs into the server.ts listeners to check polarity, seems like it has no way to access them right now. I'd keep digging, but I need to sleep and I'm not sure I'll have time to look again for a little while.