nornagon/saxi #91: errors during installation

padcrafting, 2021-09-02:
Not getting very far with installing saxi. I figured out that it needs python to be present, but it's failing at a weird error. Enclosing a screen capture

deprecated core-js@2.6.12

[image]

notmatt, 2021-11-09:
@padcrafting Not sure if you ever got past this, but in case you haven't, the core-js deprecation is just a warning and won't prevent installation. The error appears to be when building the serialport extension. At a quick guess, you might need to install/upgrade gcc or other development/build dependencies.

alexrudd2, 2023-07-01:
I ran into this problem in 2023, and believe it was actually caused by an old version of `node-gyp` in `package-lock.json`.

If you run `npm update node-gyp` first the install should work

alexrudd2, 2025-01-20:
@padcrafting  I believe this to be fixed in `0.16.0`.

If it still doesn't work for you, please open a new issue in https://github.com/alexrudd2/saxi/issues

closed 2025-01-20 by alexrudd2