nornagon/saxi #95: License Required

doctormo, 2021-11-04:
Without a defined license, your software will be All Rights Reserved by default. That means no one has permission to copy and use the software, not really. Although I think you intend the software to be open source, without a license file or file headers, there's no way to give your users the license that will protect them from copyright infringement.

GitHub will detect your license when you add the license file and automatically display it to users as a properly of the project too. Which is nice.

My personal preference (and you don't have to follow) is to use AGPL for works which can be hosted as a website: https://www.gnu.org/licenses/agpl-3.0.en.html

nornagon, 2021-11-22:
Hi, the license is already AGPL-3.0, as mentioned in the package.json file :)

nornagon, 2021-11-22:
Just to be on the safe side, I added a LICENSE file too :)

closed 2021-11-22 by nornagon