nornagon/saxi #96: Painting with Saxi

andeecollard, 2021-11-18:
First of all— this software is miraculous! I love it. 

I have been using the Axidraw branch of Watercolorbot to control my machine. https://github.com/evil-mad/wcb-ink/tree/V1_axidraw

Could you give me any advice about how to implement this style of control through Saxi? 

Are there any plans to add colour changing/brush reloading?

sorry I know it’s a niche demand.

HenrySkup, 2021-11-19:
If only to push my own use case...

I could see what was discussed in #69 being a way to create this functionality... basically can we attach meta data to either the svg document or to individual geometry elements which can affect and tweak functionality.

nornagon, 2021-11-22:
Yep, I'd love to support this sort of thing! I like what @HenrySkup said about using some sort of in-document annotations to support this rather than supporting painting actions specifically. That opens the door to other esoteric use cases, too.

andeecollard, 2021-11-22:
Thank you both for your replies. I’ve had some success on other plotters editing the gcode manually to do this sort of thing but it’s pretty random and time consuming. I don’t really understand how the Watercolorbot Axidraw Inkscape extension does what it does. 

Working with/from the svg is a clever idea.