nornagon/saxi #104: Plotting without centering

haschdl, 2022-02-20:
hello! while the scaling and placing the SCG drawing in the center of the page is a great feature, I would like to be able to plot a file "as is". My particular need is to add my signature, or the title, to a previously plotted piece. The exact placement is important in this case.
Is there a way to achieve this today?

HenrySkup, 2022-02-23:
you can use layers (actually SVG groups) to compose various elements and plot them individually and they can be in the same file.  I use this to put in crop marks

EmmaSimon, 2022-02-23:
You should also be able to uncheck "fit page" to get an SVG to be plotted to its original dimensions. But yeah layers are probably the way to go.

haschdl, 2022-02-23:
Thanks Emma! That doesn’t work as expected, Saxi seems to ignore or not be
aware of the size of the drawing. An SVG object which is placed on the
lower-right part of the canvas renders in Saxi in the middle of the screen.

nornagon, 2022-02-23:
Unchecking 'fit to page' is the way to go if you want to directly control scale and placement. It's up to you to produce an SVG of the correct dimensions in that case.

For off-center plots that still scale to the page, I usually create a layer with a rectangle that's the size of my "virtual" page, position my plot within that rectangle, and then toggle that layer off when I plot.

I'm closing this as I think this is a matter of usage, not a change to be made in saxi.

closed 2022-02-23 by nornagon

haschdl, 2022-02-25:
thanks for the tips, I agree there are multiple ways to achieve what I intended