nornagon/saxi #105: Split margin (top, bottom, left, right)

haschdl, 2022-03-06:
It is customary in photography and other forms of prints to have asymmetrical margins. A perfectly centred piece is considered less pleasing to the viewer. It is common for example to have a 10% margin in the left and right, and 7% top and 13% (or approximately double the margin size in the bottom). 

For that reason, I would like to propose to optionally have distinct margins sizes. To keep the UI exactly the same, the parsing of the margin field could be smarter, and accept both 1 number, 4 numbers (top, right, bottom, left), and accept both absolute numbers and percentages, in similar fashion as the margin attribute in CSS (see [specification](https://developer.mozilla.org/en-US/docs/Web/CSS/margin)).

For example:

- "20mm" or "20" is treated as absolute number, for all 4 sides
- "20 10 40 10" would be treated as 20mm margin top, 10mm right, 40 bottom and 10 left.
- "20%" is parsed as 20% for all sides
- "7% 10% 13% 10%" is parsed as 7% top margin, 10% right and left, and 13% bottom margin

This would also bring the advantage of having a more flexible layout system and eliminate the need for an extra "bounding box" to be added to the SVG itself.

jimmykl, 2022-03-06:
+1 for this, it's the main reason I still run artwork through Inkscape/Illustrator before plotting

ianstormtaylor, 2022-06-22:
Would also love this. For works that aren't the exact dimensions as the paper it would be useful to choose which side (top vs. bottom, left vs. right) got prioritized.