nornagon/saxi #107: Erratic pen up and down positions

haschdl, 2022-03-27:
I noticed the pen position after the plot completes, and if it's paused or stopped, it's very erratic. 

The most annoying part is that it's never possible to know if the state of the pen position in Saxi is in sync with the state in the plotter.  

In a best case scenario a second layer starts plotting with the pen too high, not touching the paper at all. The solution is just to stop, return the pen to origin, hit up, hit down, adjust pen height again, and restart.

There are situations however where the pen is at origin position, up, not touching the paper. Then I hit Plot, the pen immediately goes down and makes an ugly line from the origin to the first line. In this case the drawing is ruined.

lewi0622, 2022-08-04:
I have this issue as well. Specifically my "up height" is set lower than the default. I suspect after plotting, the pen goes to the default height, rather than the one I set. After each plot I need to press the "pen up" button to lower it to the correct height.

ostwilkens, 2022-09-12:
The issue I've had with pen height is that after the plot, when the pen has returned to home, it is lowered again and touches the paper, which results in an ink dot in the corner of each print. 
I'm now testing commenting out line 582 in planning.ts:
```js
// finally, move back to (0, 0).
motions.push(constantAccelerationPlan([curPos, {x: 0, y: 0}], profile.penUpProfile));
//motions.push(new PenMotion(Device.Axidraw.penPctToPos(penMaxUpPos), profile.penUpPos, profile.penDropDuration));
return new Plan(motions);
```
This seems to keep the pen from dropping down again. I can then click "pen up" before the next print to get it into position.