nornagon/saxi #119: documentation

benoitwimart, 2022-08-14:
Hi,

What is :
- cornering factor ?
- point joining radius ?
- path joining radius ?

HenrySkup, 2022-08-16:
While I'm not 100% positive (so @nornagon please correct me if I'm wrong):

----

**cornering factor**
: "Vertex smoothing" on paths which prevents the pen from "overshooting" the vertex due to machine's momentum.  CNC machines will do some combination of easing speed in going into and out of a vertex and "filleting" the vertex to prevent the instantaneous zero velocity point which might cause jerk.

----

These have to do with the finite precision of the plotter/material/visual perception and the trade off of efficiency.

**point joining radius**
: how close two points have to be to be considered duplicate -- to prevent "unneeded" pen up/down and/or to increase plotting efficiency 

**path joining radius** ?
: how close two path endpoints have to be to be considered/treated as the same -- to prevent "unneeded" pen up/down and/or to increase plotting efficiency

nornagon, 2022-08-16:
The above is correct. There should probably be a section of the README that explains all the various options!