nornagon/saxi #120: Wrong dimensions in the final plot

joseedil, 2022-08-27:
Hey, thanks for the great tool.

I'm using saxi on a headless raspberry pi to control an [iDraw A3 plotter](https://uunatek.com/product/idraw-handwriting-drawing-machine/). The machine uses the Axidraw firmware (currently with v2.6.5, not sure if it will work if I update it).

When using the Axidraw Inkscape plugin (v2.7.0) it correctly prints the SVG. But when using saxi I'm getting a consistent 0.8 scale factor in both X and Y directions. In the picture, it's supposed to make 10mm squares spaced 10mm from the paper border
[image].

On the saxi interface, I disabled 'fit page' and 'crop to margins'. The preview looks OK. 
[image].

I get the same 0.8 scaling factor if I use 'fit to page' with margins.
[image]

I would guess the tool is set to a different turns/mm ratio then the one on my machine.

I'm missing something?

Could you point me to where to look for in the code to help tweak it (I skimmed through it and did not locate where this kind of setting is being handled).

joseedil, 2022-08-28:
I updated the EBB firmware to v2.8.1 (latest).

Same thing.

Curiously, when I update the Inkscape plugin to v.3.5.0 (latest), it also gives the strange 0.8 scaling factor.

Surely some parameter changed from one version to the other. I'll keep investigating.

joseedil, 2022-08-28:
Ok, found the problem.

In the Inkscape plugin there is a 'native_res_factor' parameter in the [axidraw_conf.py](https://github.com/evil-mad/axidraw/blob/6f52ff308b308af1dab9f0e0bcff83daa782a538/inkscape%20driver/axidraw_conf.py) file. The default value is 1016.0 and it is used to compute the steps/in rate of the machine.

`native_res_factor = 1016.0  # Motor resolution factor, steps per inch. Default: 1016.0
# Note that resolution is defined along native (not X or Y) axes.
# Resolution is native_res_factor * sqrt(2) steps/inch in Low Resolution  (Approx 1437 steps/in)
#       and 2 * native_res_factor * sqrt(2) steps/inch in High Resolution (Approx 2874 steps/in)
`

The machine vendor provided a customized plugin. I checked it and the only thing that seems to change is this parameter which is tuned to the value `1270.0`.

Indeed, when I configure the latest inkscape plugin to this number the machine prints in the correct size. Also, the 0.8 scale factor is exactly the ratio between these two numbers.

I did not find anywhere in the saxi source code where the steps/in rate is computed to try to fix it.

joseedil, 2022-08-31:
I was able to make successful plots with this machine after changing the [stepsPerMm](https://github.com/nornagon/saxi/blob/f31840fd5747de8b5fe1c215872abf44c1aaf97b/src/planning.ts#L92) parameter to `1.25 * 5`. Not submitting a pull request as it will surely break things for Axidraw owners.

Why not expose this setting (or a machine selection setting) on the GUI?

sihil, 2023-11-21:
Just dropping in to save someone heading down the rabbit hole I've been pursuing for the last few hours: _@joseedil is talking about the original iDraw A3 and not the newer iDraw v2.0 A3._

The newer v2.0 line do not use the EBB protocol but instead use g-code. It cannot currently be hooked up to saxi. 😢 

If you have such a plotter and you're trying to use it using tools other than Inkscape then I suggest reading [this](https://github.com/bbaudry/swart-studio/blob/main/penplotting/README.md) and [this](https://github.com/gamk67/idraw2linux/blob/925073a5b550bf1b0b20225f7e604967682c622d/README.md).

thierryc, 2024-02-05:
I've delved into this matter and discovered that it's not an issue per se. I currently share a similar requirement, albeit for different reasons. I've developed a customized plotter utilizing the same EBB board. 

However, my GT2 motor pulley has 16 teeth, unlike the Axidraw's 20 teeth. 
This leads to a mathematical variation in accuracy, as I have 6.25 steps per mm compared to the Axidraw's 5 steps per mm. 
The motor itself boasts 200 steps per rotation.

In response to additional requirements, I've created a fork that hasn't been pushed as a pull request yet. WIP. In this fork, I've incorporated a dropdown menu enabling users to select the device and its associated settings. Feel free to explore it, and any feedback you may have would be greatly appreciated.

node cli.js --firmware-version
EBBv13_and_above EB Firmware Version 2.8.1

https://github.com/thierryc/saxi/tree/xylodraw

branch: xylodraw

[image]

joseedil, 2024-03-30:
I'm not aware of other clones, but this whole discussion was enlightening. 
Have been using the axidraw cli since I reported the issue, maybe it's time to try saxi again.

FS-12, 2024-07-07:
I ran into the same issue with my own built pen plotter, controlled by an EBB Board v2.7.:
Same wrong 0.8 scaling → a 100 x 100 mm box is drawn 80 x 80 mm

I edited the stepsPerMm parameter, saved the file and restarted the saxi server, but the change did not show any effect, I get the same dimensions.
Can someone give me a hint what I might be missing?

thierryc, 2024-07-08:
@FS-12 Try my version and let me know. 

https://github.com/thierryc/saxi/tree/xylodraw

Using Xylodraw in the menu should work. Count your pulley teeth.
Do you have 16 or 20 ?

FS-12, 2024-07-09:
Hi @thierryc 
I uninstalled saxi and installed your fork and it works, thanks!
You are right, I have 16 teeth GT2 pulleys.

I still don't know why changing the stepsPerMm to 6.25 with a text editor did not show any effect?! But I also dont have much experience with github and forks etc.

thierryc, 2024-07-09:
Thank you for your feedback. 

**What is the name and the version of your plotter?** (to add it into the menu)

I will continue to keep my version up to date.
My version is an experimental fork for personal Penplot work on my custom plotter (code name Xylodraw).
However, if the Saxi author is agreeable, I'd be willing to clean up the code and potentially contribute the configuration menu to the main project.

FS-12, 2024-07-09:
Its based on the [A3+ Drawing Machine](https://www.printables.com/model/102020-a3-drawing-machine-ebb-driver-board) with some tweaks I did myself and powered by the EBB board.

But I assume that the core difference between all EBB-based plotters is the size (maximum X/Y distances) and pulley teeth (GT2 16/20 teeth → scaling of the drawing influenced by [stepsPerMm](https://github.com/nornagon/saxi/blob/f31840fd5747de8b5fe1c215872abf44c1aaf97b/src/planning.ts#L92).

So instead of adding all potential pen plotter options out there it might be a better idea to allow adjustment of these core parameters?!

thierryc, 2024-07-24:
You're correct; the dropdown is just a convenient way to switch between machines. All the settings are already in place, and if you have multiple machines, it's an easy way to switch between them. However, you can simply change the stepsPerMm parameter. My SAXI fork version also includes some other custom properties.

martinnicolausson, 2024-11-27:
> I've delved into this matter and discovered that it's not an issue per se. I currently share a similar requirement, albeit for different reasons. I've developed a customized plotter utilizing the same EBB board.
> 
> However, my GT2 motor pulley has 16 teeth, unlike the Axidraw's 20 teeth. This leads to a mathematical variation in accuracy, as I have 6.25 steps per mm compared to the Axidraw's 5 steps per mm. The motor itself boasts 200 steps per rotation.
> 
> In response to additional requirements, I've created a fork that hasn't been pushed as a pull request yet. WIP. In this fork, I've incorporated a dropdown menu enabling users to select the device and its associated settings. Feel free to explore it, and any feedback you may have would be greatly appreciated.
> 
> node cli.js --firmware-version EBBv13_and_above EB Firmware Version 2.8.1
> 
> https://github.com/thierryc/saxi/tree/xylodraw
> 
> branch: xylodraw
> 
> <img alt="image" width="354" src="https://private-user-images.githubusercontent.com/197471/302451907-21517be2-04f9-4069-a2a9-98b73e04b6f4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI3MTIzODcsIm5iZiI6MTczMjcxMjA4NywicGF0aCI6Ii8xOTc0NzEvMzAyNDUxOTA3LTIxNTE3YmUyLTA0ZjktNDA2OS1hMmE5LTk4YjczZTA0YjZmNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyN1QxMjU0NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZmU0YzcwYWRkZjA2MTI2NmJiMDQyMmRmNWMyMThjNTY3MTZhMTE4MTRjNmJmZmJlMDk2ZjQ1OWEzYjE4OGJjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.an5_6tsrfg00nSERGzHXq9feA-a56mbttehc51Qe1xU">
> @joseedil the [iDraw A3 plotter](https://uunatek.com/product/idraw-handwriting-drawing-machine/). your version use probably 16 teeth pulley and a EBB board. Do you know other Axidraw Clones?

Hi,

as a Git noob, how do I go about using this repository? I've succesfully cloned it to my rpi but have no idea how to install/start this cloned version.

Thanks

alexrudd2, 2025-02-23:
> I will continue to keep my version up to date. My version is an experimental fork for personal Penplot work on my custom plotter (code name Xylodraw). However, if the Saxi author is agreeable, I'd be willing to clean up the code and potentially contribute the configuration menu to the main project.

@thierryc I have taken over from `nornagon` as maintainer, and a PR against https://github.com/alexrudd2/saxi would be welcome.

Note that we've added a bit of similar functionality to change hardware settings (for the brushless motor), but suspect your UI is an improvement.

thierryc, 2025-02-24:
Hello Alex, @alexrudd2

Thanks for the update! I’ll take care of it. 
I just need to fine-tune a few details, update my version from upstream, and then I’ll submit a PR.

I also have some improvements in mind, such as adding a button to reset to the original position and a few other enhancements.

Looking forward to contributing!

thierryc, 2025-02-24:
@martinn12345 

Hey,

Did you manage to get your project up and running? 
If you still need help, let me know where you’re stuck, and I’d be happy to guide you through the setup!

Cheers!

palvarez19, 2025-07-11:
> Hello [@palvarez19](https://github.com/palvarez19)
> 
> What hardware are you using? If you're using a "regular" AxiDraw, you don't need to use `git` at all. You can install directly from NPM. `sudo npm install -g saxi`
> 
I am using an Aliexpress plotter with an EBB board, compatible with Axidraw. I did install Saxi via NPM but thierryc's fork has modifications that I think will solve my plotting issues (plot size not accurate, same as others have mentioned above).

> For using the git version, it's unclear from your message which repository you cloned. What is the output of `git remote -v`? It should be something like this: <img alt="Image" width="863" height="77" src="https://private-user-images.githubusercontent.com/52292902/465374490-b214c5f0-90e2-42d6-898f-afd6485d583d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTIyNTQ3MTYsIm5iZiI6MTc1MjI1NDQxNiwicGF0aCI6Ii81MjI5MjkwMi80NjUzNzQ0OTAtYjIxNGM1ZjAtOTBlMi00MmQ2LTg5OGYtYWZkNjQ4NWQ1ODNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzExVDE3MjAxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNlYWZjZTI4YmJjZDBiNTdiZTliOWVjNGJhNzI0NWIyNjI1NmYxNGRhNzg5NWFhMzg3MTkwNDM2NzNkYzBhYWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7fRvT4FyuK34eGtcmAIBYXl3nHFQ1WSnNnBwwAQi7XQ">
> 
I cloned this repository:
https://github.com/thierryc/saxi/tree/xylodraw

branch: xylodraw
> From the folder you cloned into (it should be `saxi`): `npm install` and make sure there are no install errors
> 
> `npm run start`
> 
I tried this, but it doesn't seem to work. It is slow to start, and doesn't look like the fork that thierryc made. I don't know if installing Saxi via NPM is conflicting with me running the fork somehow.

Thanks for your help @alexrudd2 !

palvarez19, 2025-07-11:
> For using the git version, it's unclear from your message which repository you cloned. What is the output of `git remote -v`? It should be something like this: <img alt="Image" width="863" [...]
I thought I cloned @thierryc 's fork, but here is the output of git remote:

[image]

thierryc, 2026-11-03:
Hi @FS-12,

You helped test my Xylodraw fork a while back—thanks again for your feedback about hardware settings!

I’m now building Plot-it, a browser app for preparing artwork and pen plotting. Would you be interested in trying an early alpha?

Are you still using your custom A3+ plotter? I’d love to hear what features or settings would be most useful to you.

Thanks!
Thierry