nornagon/saxi #123: Docker/Ubuntu guide ?

666dsa666, 2022-10-18:
Hello everyone, 

Sorry for asking that but for some reason I can't get saxi to work on Ubuntu 22.04, while I might be on the right track (saxi start and tells me it's listening on port 9080) I can't seems to get to the webpage.

I would really appreciate if someone was able to share a quick guide to get it to work on Ubuntu or through docker.

[image]
[image]

lukehaas, 2022-10-29:
@666dsa666 did you try going to this address?: http://0.0.0.0:9080

666dsa666, 2022-10-29:
> @666dsa666 did you try going to this address?: http://0.0.0.0:9080

Sadly it's an headless server so i have to get to http://*ip*:9080 which doesn't seems to be working