nornagon/saxi #131: 0.15.0 is broken on RPI 4 + Axidraw A3

ostwilkens, 2022-12-14:
My plotter freezes and makes horrifying noises with the latest version. 
Not sure if it's related but looks like there is an insane amount of network traffic between my rpi and client computer during plotting. 
Things work fine in 0.14.0.

jimmykl, 2022-12-15:
+1 on this – I stopped using saxi because it sounds like it's going to trash my machine (0.15.0 on Pi 3 Model B+ with Axi SE/A3) It will draw a few lines, stutter and hung for seconds at a time but most concerningly, as above, sometimes have spasms on travel moves which sound like it'll break the machine

edouard-gillet, 2022-12-23:
I had the same issue on a Raspberry Pi 4, everything was fine in 0.14.0 and started having issues with stuttering and hanging after upgrading to 0.15.0.
I checked the different commits and figured it was most likely due to the change to drive Axidraw over WebSocket #63.
I kept 0.15.0 and rolled back that commit and it solved the issue for me.
I did not investigate further why the WebSocket change was causing stuttering issue on the Raspberry Pi.

alexrudd2, 2023-07-01:
Confirmed with `0.15.0` on an M1 Mac.

The individual commands to the pen up/down work fine, but trying to plot sends the motors into horrible spasms until I pull the plug.

alexrudd2, 2023-07-01:
OK the grinding improves with rehoming, but there is still something strange happening.  There are a huge amount of connect and disconnect messages (when killing the server) as compared to `0.14.0`

`0.14.0`: Notice no duplicate "connect" messages, and the "disconnected" messages are every 5s.
[image]

`0.15.0`: Notice the duplicate "connect" messages (varies), and the "disconnected" messages occur many times per second.
[image]

julienes, 2023-09-19:
I've just reinstalled version 0.15, which I had uninstalled last time because of this problem, and I see that it's still there. I've just installed it to be able to use the new brushless servo. Any news about this problem?

alexrudd2, 2023-09-19:
`0.15` doesn't include code for the brushless yet, and the dev is pretty busy with a newborn right now.

You can try installing from one of the forks, where we're experimenting before trying to merge here.
- https://github.com/alexrudd2/saxi/tree/brushless
- https://github.com/jedahan/saxi/tree/brushless-v2

julienes, 2023-09-19:
Oh ok, I'll take a look. I did that https://github.com/nornagon/saxi/issues/136 and now I can control my brushless with Saxi. But I have the problem of the Axidraw freezing during the plot

oskay, 2023-09-19:
> Oh ok, I'll take a look. I did that #136 and now I can control my brushless with Saxi. But I have the problem of the Axidraw freezing during the plot

You might be able to revert to the earlier Saxi version that wasn't freezing for you (if there is one), and then "manually" apply the patches from #136.

julienes, 2023-09-20:
Thank you! I've changed it back to version 0.14.0. I tried yesterday but it didn't seem to work when I changed the files to the new ones. I changed the lines manually and it works. Glad to be able to continue using Saxi.

jedahan, 2023-10-01:
@osresearch i think this is it

jedahan, 2023-10-01:
Regarding the many listeners, i wonder if these lines shoule be ported https://github.com/nornagon/saxi/commit/593c451e8ba252b1e85f27c373df8c2aa621b723?diff=split#diff-8a8ae07582c9d433ec8c2e5c4310ff8901e604f4965c5b90a49117ad46c47595L236-L247

jedahan, 2023-10-01:
I manually 'reverted' the commit with webserial, and things are working smoothly.

Would be great to get webserial back in and working at a future point.

closed 2023-10-26 by nornagon