nornagon/saxi #134: Build errors

SimonVanherweghe, 2023-05-09:
How can I build this?

I installed typescript as a local dev dependency `v^4.9.5`
I'm running node `v18.16.0`

This is what I get after running `npm run prepare`

```bash
> saxi@0.14.1 prepare
> rimraf dist && npm run build

> saxi@0.14.1 prebuild
> npm run lint

> saxi@0.14.1 lint
> eslint --cache --ext .ts,.tsx src

> saxi@0.14.1 build
> tsc && webpack --mode=production

src/serialport-serialport.ts:21:14 - error TS2720: Class 'SerialPortSerialPort' incorrectly implements class 'SerialPort'. Did you mean to extend 'SerialPort' and inherit its members as a subclass?
  Property 'forget' is missing in type 'SerialPortSerialPort' but required in type 'SerialPort'.

21 export class SerialPortSerialPort extends EventEmitter implements SerialPort {
                ~~~~~~~~~~~~~~~~~~~~

  node_modules/@types/w3c-web-serial/index.d.ts:56:5
    56     forget(): Promise<void>;
           ~~~~~~
    'forget' is declared here.

src/server.ts:241:3 - error TS2741: Property 'forget' is missing in type 'SerialPortSerialPort' but required in type 'SerialPort'.

241   return port
      ~~~~~~~~~~~

  node_modules/@types/w3c-web-serial/index.d.ts:56:5
    56     forget(): Promise<void>;
           ~~~~~~
    'forget' is declared here.

src/server.ts:290:20 - error TS2345: Argument of type 'SerialPortSerialPort' is not assignable to parameter of type 'SerialPort'.

290     return new EBB(new SerialPortSerialPort(path));
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/server.ts:294:22 - error TS2345: Argument of type 'SerialPortSerialPort' is not assignable to parameter of type 'SerialPort'.

294       return new EBB(new SerialPortSerialPort(ebbs[0]));
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Found 4 errors in 2 files.

Errors  Files
     1  src/serialport-serialport.ts:21
     3  src/server.ts:241
```

alexrudd2, 2023-07-21:
@SimonVanherweghe  Does this still occur for you from `main`?  It also failed to build for me, but we got a lot of changes merged that improved things.

SimonVanherweghe, 2023-08-17:
I merged the main, accepted all incoming changes, but still the same errors. Is there a dependency missing?

alexrudd2, 2023-08-21:
Huh, works for me with both `v18.17.1` and `20.5.1`.  What if you try removing all of `node_modules` and then running `npm install .`?  That installs all the current dependencies and then tries to compile.

SimonVanherweghe, 2023-08-24:
After cloning this repo again, it works now. Thanks for the help

closed 2023-08-24 by SimonVanherweghe