nornagon/saxi #135: PayloadTooLargeError: request entity too large

Monokai, 2023-05-21:
I'm trying to plot an .SVG file of 9MB with lots of tiny paths. Saxi runs on a Raspberry PI with 4GB memory. The path planning phase goes ok, but nothing happens.

I'm running the latest Saxi on Node 18.16 on the PI. Looking in `/var/log/syslog` I see this:

```log
PayloadTooLargeError: request entity too large
at readStream (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/raw-body/index.js:156:17)
at getRawBody (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/raw-body/index.js:109:12)
at read (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/body-parser/lib/read.js:79:3)
at jsonParser (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/body-parser/lib/types/json.js:135:5)
at Layer.handle [as handle_request] (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/express/lib/router/layer.js:95:5)
at trim_prefix (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/express/lib/router/index.js:328:13)
at /home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/express/lib/router/index.js:286:9
at Function.process_params (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/express/lib/router/index.js:346:12)
at next (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/express/lib/router/index.js:280:10)
at serveStatic (/home/pi/.nvm/versions/node/v18.16.0/lib/node_modules/saxi/node_modules/serve-static/index.js:75:16)

```

Any ideas?

wwatkns, 2024-07-18:
Hey Monokai!

I had the same issue, I read through this thread on Stack Overflow https://stackoverflow.com/questions/19917401/error-request-entity-too-large then changed the line 19 in `/src/server.ts` from
```javascript
app.use(express.json({limit: maxPayloadSize}));
``` 
to
```javascript
let bodyParser = require('body-parser');
app.use(bodyParser.json({ limit: maxPayloadSize }));
app.use(bodyParser.urlencoded({ limit: maxPayloadSize,  extended: true, parameterLimit: 1000000 }));
```
I don't know the exact reason but it works for me now.
It seems to be an issue connected to the number of small lines rather than the file size.

Monokai, 2024-07-18:
Hah! Nice yo see you here @wwatkns, and thanks :) Now if I only remember what I was plotting a year ago. I might even have solved it by truncating all floating points, but your solution looks more future proof.

wwatkns, 2024-07-20:
Never mind, the issue popped up again for a different plot after I slowed down the `down acc` from 200 to 150mm/s²