nornagon/saxi #139: tests for EBB are broken

alexrudd2, 2023-06-27:
```
> saxi@0.14.0 test
> jest

 PASS  src/__tests__/planning.test.ts
 FAIL  src/__tests__/ebb.test.ts
  ● Test suite failed to run

    src/__tests__/ebb.test.ts:23:22 - error TS2339: Property 'list' does not exist on type 'typeof EBB'.

    23     expect(await EBB.list()).toEqual([])
                            ~~~~
    src/__tests__/ebb.test.ts:28:22 - error TS2339: Property 'list' does not exist on type 'typeof EBB'.

    28     expect(await EBB.list()).toEqual([])
```

Here are the lines that broke it:
https://github.com/nornagon/saxi/commit/593c451e8ba252b1e85f27c373df8c2aa621b723#diff-290f6b1d4ab368bc49cff4217841f3c2ac88b2c7c98b421a81844b7e26d1aa22L19

alexrudd2, 2023-06-27:
Additionally, the use of `TransformStream` won't work in Node.js since that's using a browser Streams API.

https://github.com/nornagon/saxi/commit/593c451e8ba252b1e85f27c373df8c2aa621b723#diff-f6bf3f98cc0ece205255773b4b55b5b586ba90bfc083d9cb5d1b91f9520eb63cR1

nornagon, 2023-06-27:
Yeah, I'd noticed this too but didn't have time to fix it. I'd be happy to review a PR!

alexrudd2, 2023-06-28:
> Yeah, I'd noticed this too but didn't have time to fix it. I'd be happy to review a PR!

I've been starting to learn this codebase (and TypeScript, actually) in my fork: https://github.com/alexrudd2/saxi

When I figure some things out I'd be happy to send some PRs your way, or you can cherry-pick if you'd like.  To start, I'm trying to upgrade to `node-serialport` v10 (which is tricky) and upgrade the other dependencies (pretty easy).  I'll get a real Axidraw hardware soon to test.

alexrudd2, 2023-06-28:
> Additionally, the use of `TransformStream` won't work in Node.js since that's using a browser Streams API.

This can be fixed by adding the following to `src/regex-transform-stream.ts`:
`import { TransformStream } from "web-streams-polyfill/ponyfill";`

alexrudd2, 2023-06-28:
OK, with some more experimentation, and reading https://github.com/nornagon/saxi/issues/131, I actually think reverting all the WebSerial stuff is the way to go.  It's clearly broken.

closed 2023-07-10 by nornagon