nornagon/saxi #177: Installation on a Pi Zero fails due to unsupported ARMv6 architecture

osresearch, 2023-07-29:
The version of node.js mentioned in the README is very old -- 12.x has been deprecated for a while and the current recommended one is 20.5 (LTS is 18.17).  In either case they are not built for the ARMv6 in the Pi Zero and result in an error message:
```
% curl -sL https://deb.nodesource.com/setup_lts.x | sudo -E bash -

## Installing the NodeSource Node.js 18.x repo...

## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js 4 and later.
```

It looks like the last `linux-armv6l` tarball available is 11.8.0: https://nodejs.org/dist/v11.8.0/node-v11.8.0-linux-armv6l.tar.xz  I've instead successfully installed npm from the Pi's package manager with `sudo apt install npm`, which is currently 7.5.2.

jedahan, 2023-08-07:
Running the latest Raspberry Pi OS Lite (32-bit) for Pi Zero W v1.1, I just used upstream packages (which installs node 12) like so:

```shell
sudo apt-get install --yes nodejs npm

mkdir -p ~/.config/npm/global
npm config set prefix '~/.config/npm/global'
echo 'export NPM_CONFIG_PREFIX=~/.config/npm/global' >> ~/.profile  # maybe unnecessary
. ~/.profile

npm install --global saxi
```

Then, using systemd I just put the full path in the `ExecStart` line (`ExecStart=/home/jedahan/.npm/global/bin/saxi`)

jedahan, 2023-09-01:
I've since switched to using https://github.com/nodejs/unofficial-builds for armv6l

jedahan, 2023-09-06:
And I've got a PR #186 for making `npm build` work on pi zero