nornagon/saxi #178: Leave pen up at end of plot

osresearch, 2023-07-29:
It seems that after a successful plot, the carriage returns to home and executes a pen down. Is there a way to turn this off so that it doesn't mark the paper?  I'm not sure why it does this, since the code in function `plan` seems to end with a max-up pen position:

```
  // finally, move back to (0, 0).
  motions.push(constantAccelerationPlan([curPos, {x: 0, y: 0}], profile.penUpProfile));
  motions.push(new PenMotion(Device.Axidraw.penPctToPos(penMaxUpPos), profile.penUpPos, profile.penDropDuration));
  return new Plan(motions);
```

unless it is being modified in `withPenHeights()`? I'm not clear on exactly what is happening here, although it seems odd that the last motion is being treated special and overriding the actual command:
```
                if (j === this.motions.length - 3) {
                    return new PenMotion(penDownHeight, exports.Device.Axidraw.penPctToPos(0), motion.duration());
                }
                else if (j === this.motions.length - 1) {
                    return new PenMotion(exports.Device.Axidraw.penPctToPos(0), penUpHeight, motion.duration());
                }
```

(I'm running this on a Pi Zero with the `saxi` version 0.15.0 installed via npm)

jedahan, 2023-08-09:
I am not observing this behavior - sounds dumb but could you share an SVG that does this and I can try it out?

lostPixels, 2023-10-05:
Currently experiencing this on two Axidraws. It never used to happen. Please let me know if I can help with debugging info.

judbd, 2023-12-11:
I have this behavior on my Axidraw SE/A3 too.

morozgrafix, 2024-10-19:
Just installed it on Raspberry Pi and confirming that pen goes down at the end of the plot.

alexrudd2, 2025-02-16:
If I had to guess, there is a difference between the `WebSerial` and Node `SerialPort` implementations.  (but I can't find it yet)