nornagon/saxi #194: Saxi becomes unresponsive while plotting, and Axidraw starts/stops too

lostPixels, 2023-10-14:
I'm running Saxi remotely on a RPI and experiencing some odd behavior when plotting. The UI becomes unresponsive, and the Axidraw sputters. The oddest thing is that the machine will return to normal function if the UI is closed. This has led me to quickly closing the tab once I hear the stepper engage, which doesnt always eliminate the possibility that the steppers will sputter and lose alignment. 

Ive experienced this with many different SVGs of various complexities. 

In the attached image you can see the http response time grow to be quite large. 

Also included is a video of the Axidraw as it sputters: https://imgur.com/a/pq8US58

Saxi version: 0.15.0
Node Version: v18.16.1
Using NVM: Yes
Raspberry Pi version: 4
Axidraw: V3/A3 (newest Firmware)

Plot File: 20mb~ SVG

[image]

jedahan, 2023-10-14:
As a sanity check, i looked at the length of a serialized plan for the following svg.

svg is 800kb, `JSON.stringify().length / 1024 / 1024` = `9.26 Megabytes`

I generally expect the rendered paths to be much larger than the svg (vector to raster, all that jazz), but this seems a bit excessive.

If its a matter of timing and blocking on the server, there might be simple fixes, like enabling gzip support.

If its a memory usage thing, we might want to try more compact representations.

This might be a red herring, but its a start.

[image]

jedahan, 2023-10-14:
Note that I have experienced this issue, and was able to stop it completely by removing WebSerial support from the repo. It would be ideal to fix webserial though, see alexrudd2/saxi#89

alexrudd2, 2023-10-14:
I think this is the same as https://github.com/nornagon/saxi/issues/131

It's harder to reproduce on more powerful hardware, but most people on RPi's usually see it intermittently

alexrudd2, 2023-10-16:
@lostPixels 

Can you try the code in this pull request: https://github.com/alexrudd2/saxi/pull/113 and see if it helps any?

lostPixels, 2023-10-16:
Hey @alexrudd2 , I spent a while this morning getting the dev version on to my RPI but couldnt get it to run, I kept getting `Error: listen EADDRINUSE: address already in use :::9080` 

I can report though that moving back down to 0.13.0 had resolved my glitching issue for now.

jedahan, 2023-10-18:
Next steps might be to figure out how to profile memory usage after loading a plan, and then see what changes help.

Following https://nodejs.org/en/docs/guides/diagnostics/memory .

closed 2023-10-26 by nornagon