nornagon/saxi #203: Extra thick stroke width in FireFox

dtgreene, 2023-11-23:
Hello 👋 I've noticed that paths rendered in FireFox appear extra wide when compared to Chrome.  Here are some screenshots to show what I mean:

FireFox:
[image]

Chrome:
[image]

I found that adding this CSS rule fixed the issue for me:
```
path {
  vector-effect: non-scaling-stroke;
}
```

alexrudd2, 2023-11-27:
[image]
I cannot reproduce this.

I see the page sizes are different.  Are you sure the `Visualized Stroke width (mm)` is the same on both browsers?  Can you try with the same settings on both browsers?

alexrudd2, 2023-11-27:
> Ah yeah you're right, it was the page size. Turns out the smaller page sizes causes the stroke width to appear thicker and I didn't notice they were different when making the issue, my bad.

Glad you figured it out.  If you're interested in playing with the code more, I've got a more [active fork](https://github.com/alexrudd2/saxi) where we experiment before submitting PR's here.

dtgreene, 2023-11-27:
Oh awesome! I'll try that out later tonight